esp-rs/std-training #55: Cannot find macro 'llvm_asm' in this scope

reinhardbluelab, 2022-04-22:
I am following the guide [here](https://espressif-trainings.ferrous-systems.com/02_4_hello_board.html) and when trying to run the _hardware-check_ example:
`cargo espflash --release --monitor /dev/tty.usbserial-1130`

I get the following error:
`error: could not compile "riscv" due to 366 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
error: cannot find macro "llvm_asm" in this scope`

`error[E0635]: unknown feature "llvm_asm"
  --> /Users/reinhard/.cargo/registry/src/github.com-1ecc6299db9ec823/riscv-0.7.0/src/lib.rs:17:45
   |
17 | #![cfg_attr(feature = "inline-asm", feature(llvm_asm))]`

I have tried pinning nightly to nightly-2022-03-10 (using `rustup install nightly-2022-03-10` and `rustup component add rust-src --toolchain nightly-2022-03-10`) and to 2022-04-21, but both result in errors.

justahero, 2022-04-22:
Hello @reinhardbluelab ,

could you check out branch `fix-riscv-dependency` and see if that works for you? The branch contains fixes to the dependencies.

closed 2022-04-22 by justahero

marcelbuesing, 2022-04-22:
> Hello @reinhardbluelab ,
> 
> could you check out branch `fix-riscv-dependency` and see if that works for you? The branch contains fixes to the dependencies.

I had the same issue and switching to the branch fixed it. I had to delete the `~\.espressif` folder though and had a issue with libclang not being found. The following worked for me on arch. I assume it might be sufficient to just switch to the branch on other systems.

```
LIBCLANG_PATH=/usr/lib/libclang.so cargo build --target riscv32imc-esp-espidf
```

justahero, 2022-04-22:
The fix is in the `main` branch now.

In case the issue still persists, please don't hesitate to reopen this ticket again.

justahero, 2022-04-22:
> [...]
> 
> ```
> LIBCLANG_PATH=/usr/lib/libclang.so cargo build --target riscv32imc-esp-espidf
> ```
> 
> Update: Just noticed an old LIBCLANG_PATH config in my zshrc pointing to an incorrect location.

That is a good point to look out for.

reinhardbluelab, 2022-04-25:
> Hello @reinhardbluelab ,
> 
> could you check out branch `fix-riscv-dependency` and see if that works for you? The branch contains fixes to the dependencies.

The fix works fine. Thank you.

PR #56 (2022-04-22): Use `rev` option to set fixed commit hash
This change sets a fixed commit revision to the `riscv` patch annotation in all `Cargo.toml` files.

Using `master` as a `branch` option resulted in different commit hashes at different times, which occurred due to changes in the [`riscv` crate](https://github.com/rust-embedded/riscv). This crate has seen some changes in the last days / weeks that broke the current setup, therefore a working revision is set.

All examples compiled successfully but have not been tested with the hardware itself.

This should fix #55.