esp-rs/std-training #57: intro/http-client/example doesn't build without updating riscv to 0.8.0 in esp32-c3-dkc02-bsc

galamdring, 2022-04-22:
when trying to run the http-client example

    cargo doc --open

I got a multitude of errors about the llvm_asm! macro

    error: cannot find macro `llvm_asm` in this scope

I was able to resolve it by downloading the [esp-pacs](https://github.com/esp-rs/esp-pacs) repo and patch esp32c3 in common/lib/esp32-c3-dkc01-bsc/Cargo.toml

    [dependencies]
    esp32c3 = "=0.4.0"
    ...
    [patch.crates.io]
    esp32c3 = { path = "/Users/lmckechnie/esp/esp-pacs/esp32c3" }

This updated the riscv dependency to 0.8.0. I also added the riscv 0.8.0 dependency to intro/http-client/exercise/Cargo.toml

justahero, 2022-04-22:
Hello @galamdring ,

thanks for the report! It seems you ran into the same issue as reported in #55. A fix has been published earlier today to the `main` branch, where the `riscv` dependencies have been set to a fixed revision.

Could you update the repository & check the latest version in the `main` branch works for you?

galamdring, 2022-04-22:
@justahero Thank you, that did fix my problem. I thought I had searched here for the error, not sure how i missed that one.

closed 2022-04-22 by galamdring