esp-rs/std-training #76: Can I translate the book into Chinese?

YuaXan, 2022-07-05:
For title.

Narukara, 2023-02-02:
I am translating now.

here: https://github.com/Narukara/espressif-trainings-cn

SergioGasquez, 2023-03-30:
Hi! The book has received a major update, if the Chinese translation is up-to-date we could try to review it and include it officially if you are okay with that @Narukara.

I think it would be an awesome resource to have! Thanks for your efforts on translating!

Narukara, 2023-03-30:
I'm okay with that. But the translation is not yet complete. This will take a little longer because I am currently busy with my graduation project.😢 I will be back in two months.

SergioGasquez, 2023-03-31:
Of course, no worries. It should also give us time to curate it a bit more!

SergioGasquez, 2023-06-01:
I just noticed that you resumed the translation @Narukara, thanks for all your efforts, do you think we can include it in this repo once its finished?

SergioGasquez, 2023-06-22:
Hi! Again, thanks for all your efforts on translating! I see that your fork is updated (only missing the content a few minor commits) do you think we are ready to add that translations section and point to your book?

Narukara, 2023-06-23:
Yes, I think we are ready. I'll probably finish the rest of my book within a few days.

closed 2023-06-27 by MabezDev