esp-rs/std-training #81: Update board refrences

SergioGasquez, 2022-07-20:
There are some references for ESP32-C3-DevKitC-02 in the repository that might need to be updated.
- BSC is actually using Rust board pins but the name suggest is for ESP32-C3-DevKitC-02.
  - It may be worth to add some comment [like in other parts](https://github.com/ferrous-systems/espressif-trainings/blob/3d1612b24d09b579884ae21809fbc96e93871237/advanced/i2c-sensor-reading/solution/src/main_both.rs#L28-L30) for users that are actually using ESP32-C3-DevKitC-02
- [`intro/hardware-check` README](https://github.com/ferrous-systems/espressif-trainings/tree/main/intro/hardware-check)

SergioGasquez, 2022-11-21:
Hi @Mirabellensaft! Is there any reason to close the issue?

I think it might be misleading to readers, as the [BSC](https://github.com/ferrous-systems/espressif-trainings/tree/3d1612b24d09b579884ae21809fbc96e93871237/common/lib/esp32-c3-dkc02-bsc) and the other references to  ESP32-C3-DevKitC-02 actually mean ESP32-C3-DevKit-RUST-1, and all the pins are configured for that board.

Mirabellensaft, 2022-11-21:
Hi @SergioGasquez, I thought this was resolved, but I'll reopen it and double check tomorrow.

closed 2023-03-17 by SergioGasquez