esp-rs/std-training #93: hardware-check: rv_utils.h: invalid input constraint 'rK' in asm

doxxx, 2022-08-25:
I have the ESP32-C3-Devkit-RUST-1 board and I'm trying to follow the Embedded Rust on Espressif book. I'm at the point where I'm trying to compile the hardware-check application, but it's failing with these errors:

```
<lots more output snipped>

  Using custom user-supplied esp-idf repository at '/Users/gordon/esp/esp-idf' (detected from env variable `IDF_PATH`)
  Using esp-idf v5.1.0 at '/Users/gordon/esp/esp-idf'
  Output: "/Users/gordon/dev/esp32-rust/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-d1a5f1a58ff63106/out/bindings.rs"
  Bindgen builder flags: ["/Users/gordon/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.5/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DUNITY_INCLUDE_CONFIG_H", "-I/Users/gordon/esp/esp-idf/components/riscv/include", "-I/Users/gordon/dev/esp32-rust/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-d1a5f1a58ff63106/out/build/config", "-I/Users/gordon/esp/esp-idf/components/newlib/platform_include", "-I/Users/gordon/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include", "-I/Users/gordon/esp/esp-idf/components/freertos/esp_additions/include/freertos", "-I/Users/gordon/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include", "-I/Users/gordon/esp/esp-idf/components/freertos/esp_additions/include", "-I/Users/gordon/esp/esp-idf/components/esp_hw_support/include", "-I/Users/gordon/esp/esp-idf/components/esp_hw_support/include/soc", "-I/Users/gordon/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3", "-I/Users/gordon/esp/esp-idf/components/esp_hw_support/port/esp32c3/.", "-I/Users/gordon/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include", "-I/Users/gordon/esp/esp-idf/components/heap/include", "-I/Users/gordon/esp/esp-idf/components/log/include", "-I/Users/gordon/esp/esp-idf/components/soc/include", "-I/Users/gordon/esp/esp-idf/components/soc/esp32c3/.", "-I/Users/gordon/esp/esp-idf/components/soc/esp32c3/include", "-I/Users/gordon/esp/esp-idf/components/hal/esp32c3/include", "-I/Users/gordon/esp/esp-idf/components/hal/include", "-I/Users/gordon/esp/esp-idf/components/hal/platform_port/include", "-I/Users/gordon/esp/esp-idf/components/esp_rom/include", "-I/Users/gordon/esp/esp-idf/components/esp_rom/include/esp32c3", "-I/Users/gordon/esp/esp-idf/components/esp_rom/esp32c3", "-I/Users/gordon/esp/esp-idf/components/esp_common/include", "-I/Users/gordon/esp/esp-idf/components/esp_system/include", "-I/Users/gordon/esp/esp-idf/components/esp_system/port/soc", "-I/Users/gordon/esp/esp-idf/components/esp_system/port/include/riscv", "-I/Users/gordon/esp/esp-idf/components/esp_system/port/include/private", "-I/Users/gordon/esp/esp-idf/components/lwip/include", "-I/Users/gordon/esp/esp-idf/components/lwip/include/apps", "-I/Users/gordon/esp/esp-idf/components/lwip/include/apps/sntp", "-I/Users/gordon/esp/esp-idf/components/lwip/lwip/src/include", "-I/Users/gordon/esp/esp-idf/components/lwip/port/esp32/include", "-I/Users/gordon/esp/esp-idf/components/lwip/port/esp32/include/arch", "-I/Users/gordon/esp/esp-idf/components/esp_ringbuf/include", "-I/Users/gordon/esp/esp-idf/components/efuse/include", "-I/Users/gordon/esp/esp-idf/components/efuse/esp32c3/include", "-I/Users/gordon/esp/esp-idf/components/driver/include", "-I/Users/gordon/esp/esp-idf/components/driver/deprecated", "-I/Users/gordon/esp/esp-idf/components/esp_pm/include", "-I/Users/gordon/esp/esp-idf/components/mbedtls/port/include", "-I/Users/gordon/esp/esp-idf/components/mbedtls/mbedtls/include", "-I/Users/gordon/esp/esp-idf/components/mbedtls/mbedtls/library", "-I/Users/gordon/esp/esp-idf/components/esp_app_format/include", "-I/Users/gordon/esp/esp-idf/components/bootloader_support/include", "-I/Users/gordon/esp/esp-idf/components/bootloader_support/bootloader_flash/include", "-I/Users/gordon/esp/esp-idf/components/app_update/include", "-I/Users/gordon/esp/esp-idf/components/spi_flash/include", "-I/Users/gordon/esp/esp-idf/components/pthread/include", "-I/Users/gordon/esp/esp-idf/components/esp_timer/include", "-I/Users/gordon/esp/esp-idf/components/app_trace/include", "-I/Users/gordon/esp/esp-idf/components/esp_event/include", "-I/Users/gordon/esp/esp-idf/components/nvs_flash/include", "-I/Users/gordon/esp/esp-idf/components/esp_phy/include", "-I/Users/gordon/esp/esp-idf/components/esp_phy/esp32c3/include", "-I/Users/gordon/esp/esp-idf/components/vfs/include", "-I/Users/gordon/esp/esp-idf/components/esp_netif/include", "-I/Users/gordon/esp/esp-idf/components/wpa_supplicant/include", "-I/Users/gordon/esp/esp-idf/components/wpa_supplicant/port/include", "-I/Users/gordon/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include", "-I/Users/gordon/esp/esp-idf/components/esp_wifi/include", "-I/Users/gordon/esp/esp-idf/components/unity/include", "-I/Users/gordon/esp/esp-idf/components/unity/unity/src", "-I/Users/gordon/esp/esp-idf/components/cmock/CMock/src", "-I/Users/gordon/esp/esp-idf/components/console", "-I/Users/gordon/esp/esp-idf/components/http_parser", "-I/Users/gordon/esp/esp-idf/components/esp-tls", "-I/Users/gordon/esp/esp-idf/components/esp-tls/esp-tls-crypto", "-I/Users/gordon/esp/esp-idf/components/esp_adc/include", "-I/Users/gordon/esp/esp-idf/components/esp_adc/interface", "-I/Users/gordon/esp/esp-idf/components/esp_adc/esp32c3/include", "-I/Users/gordon/esp/esp-idf/components/esp_adc/deprecated/include", "-I/Users/gordon/esp/esp-idf/components/esp_eth/include", "-I/Users/gordon/esp/esp-idf/components/esp_gdbstub/include", "-I/Users/gordon/esp/esp-idf/components/esp_gdbstub/riscv", "-I/Users/gordon/esp/esp-idf/components/esp_gdbstub/esp32c3", "-I/Users/gordon/esp/esp-idf/components/esp_hid/include", "-I/Users/gordon/esp/esp-idf/components/tcp_transport/include", "-I/Users/gordon/esp/esp-idf/components/esp_http_client/include", "-I/Users/gordon/esp/esp-idf/components/esp_http_server/include", "-I/Users/gordon/esp/esp-idf/components/esp_https_ota/include", "-I/Users/gordon/esp/esp-idf/components/esp_lcd/include", "-I/Users/gordon/esp/esp-idf/components/esp_lcd/interface", "-I/Users/gordon/esp/esp-idf/components/protobuf-c/protobuf-c", "-I/Users/gordon/esp/esp-idf/components/protocomm/include/common", "-I/Users/gordon/esp/esp-idf/components/protocomm/include/security", "-I/Users/gordon/esp/esp-idf/components/protocomm/include/transports", "-I/Users/gordon/esp/esp-idf/components/esp_local_ctrl/include", "-I/Users/gordon/esp/esp-idf/components/esp_psram/include", "-I/Users/gordon/esp/esp-idf/components/sdmmc/include", "-I/Users/gordon/esp/esp-idf/components/esp_serial_slave_link/include", "-I/Users/gordon/esp/esp-idf/components/espcoredump/include", "-I/Users/gordon/esp/esp-idf/components/espcoredump/include/port/riscv", "-I/Users/gordon/esp/esp-idf/components/wear_levelling/include", "-I/Users/gordon/esp/esp-idf/components/fatfs/diskio", "-I/Users/gordon/esp/esp-idf/components/fatfs/vfs", "-I/Users/gordon/esp/esp-idf/components/fatfs/src", "-I/Users/gordon/esp/esp-idf/components/idf_test/include", "-I/Users/gordon/esp/esp-idf/components/idf_test/include/esp32c3", "-I/Users/gordon/esp/esp-idf/components/ieee802154/include", "-I/Users/gordon/esp/esp-idf/components/json/cJSON", "-I/Users/gordon/esp/esp-idf/components/mqtt/esp-mqtt/include", "-I/Users/gordon/esp/esp-idf/components/spiffs/include", "-I/Users/gordon/esp/esp-idf/components/wifi_provisioning/include", "--sysroot=/Users/gordon/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../riscv32-esp-elf", "-I/Users/gordon/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../riscv32-esp-elf/include", "-x", "c", "-DESP_IDF_COMP_BOOTLOADER_SUPPORT_ENABLED", "-DESP_IDF_COMP_IEEE802154_ENABLED", "-DESP_IDF_COMP_UNITY_ENABLED", "-DESP_IDF_COMP_ESP_PHY_ENABLED", "-DESP_IDF_COMP_WEAR_LEVELLING_ENABLED", "-DESP_IDF_COMP_FATFS_ENABLED", "-DESP_IDF_COMP_VFS_ENABLED", "-DESP_IDF_COMP_PERFMON_EN [... cut ...]
  /Users/gordon/esp/esp-idf/components/riscv/include/riscv/rv_utils.h:152:9: error: invalid input constraint 'rK' in asm
  /Users/gordon/esp/esp-idf/components/riscv/include/riscv/rv_utils.h:155:9: error: invalid input constraint 'rK' in asm
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

I'm using macOS 10.15.7 on an Intel MacBook. I installed the software prerequisites as described in the book using `rustup` and `brew`. I installed `esp-idf` v5.1.0 from git master and was able to successfully compile and flash the C hello_world application from `esp-idf` onto my device.

doxxx, 2022-08-25:
I've also tried building without `esp-idf` 5.1.0 activated in my shell, and having removed `~/.espressif` entirely. I get a very similar error, just in a different header file:

```
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Tag("v4.4.1") }
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1'...
  Note: checking out '1329b19fe494500aeb79d19b27cfd99b40c37aec'.

  You are in 'detached HEAD' state. You can look around, make experimental
  changes and commit them, and you can discard any commits you make in this
  state without impacting any branches by performing another checkout.

  If you want to create a new branch to retain commits you create, you may
  do so (now or later) by using -b with the checkout command again. Example:

    git checkout -b <new-branch-name>

Checking out files: 100% (9635/9635), done.
  Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
  Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
  Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib_esp32'
  Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c3_family'
  Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
  Submodule 'components/cbor/tinycbor' (https://github.com/intel/tinycbor.git) registered for path 'components/cbor/tinycbor'
  Submodule 'components/cmock/CMock' (https://github.com/ThrowTheSwitch/CMock.git) registered for path 'components/cmock/CMock'
  Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
  Submodule 'components/esp_phy/lib' (https://github.com/espressif/esp-phy-lib.git) registered for path 'components/esp_phy/lib'
  Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
  Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
  Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
  Submodule 'components/ieee802154/lib' (https://github.com/espressif/esp-ieee802154-lib.git) registered for path 'components/ieee802154/lib'
  Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
  Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
  Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
  Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
  Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
  Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
  Submodule 'components/openthread/lib' (https://github.com/espressif/esp-thread-lib.git) registered for path 'components/openthread/lib'
  Submodule 'components/openthread/openthread' (https://github.com/espressif/openthread.git) registered for path 'components/openthread/openthread'
  Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
  Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
  Submodule 'components/tinyusb/tinyusb' (https://github.com/espressif/tinyusb.git) registered for path 'components/tinyusb/tinyusb'
  Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
  Submodule 'examples/build_system/cmake/import_lib/main/lib/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
  Submodule 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' (https://github.com/espressif/esp-cryptoauthlib.git) registered for path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/asio/asio'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/bt/controller/lib_esp32'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/bt/controller/lib_esp32c3_family'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/bt/host/nimble/nimble'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/cbor/tinycbor'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/cmock/CMock'...
  Cloning into '/Users/gordon/.espressif/esp-idf/v4.4.1/components/coap/libcoap'...
[... cut ...]
  /Users/gordon/.espressif/esp-idf/v4.4.1/components/hal/esp32c3/include/hal/cpu_ll.h:118:9: error: invalid input constraint 'rK' in asm
  /Users/gordon/.espressif/esp-idf/v4.4.1/components/hal/esp32c3/include/hal/cpu_ll.h:120:5: error: invalid input constraint 'rK' in asm
  /Users/gordon/.espressif/esp-idf/v4.4.1/components/hal/esp32c3/include/hal/cpu_ll.h:126:5: error: invalid input constraint 'rK' in asm
  /Users/gordon/.espressif/esp-idf/v4.4.1/components/hal/esp32c3/include/hal/cpu_ll.h:127:5: error: invalid input constraint 'rK' in asm
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

SergioGasquez, 2022-09-08:
Hello! For the training, [we use tag `v4.4.1`](https://github.com/ferrous-systems/espressif-trainings/blob/main/intro/hardware-check/.cargo/config.toml#L31) of esp-idf, please follow [the directions in the book](https://espressif-trainings.ferrous-systems.com/02_2_software.html#troubleshooting) for a clean installation and make sure you do a `cargo clean` when building with the new esp-idf installed.

davidcornu, 2022-09-15:
@doxxx I ran into the same issue and managed to resolve it by doing the following:

1. Following the instructions [here](https://espressif-trainings.ferrous-systems.com/02_2_software.html#troubleshooting)<br>(the formatting confused me initially and it took me way too long to realize that everything between `3.` and `4.` should be treated as if it was part of `3.` 🤦‍♂️)<br><br>[image]<br>This netted me with a clone of `esp-idf` at the `v4.4.1` at `~/esp/esp-idf`.
2. In order to get the `esp-idf-sys` crate to use that directory and not `~/.espressif/...` I needed to `export IDF_PATH=~/esp/esp-idf/`
3. In order to get `bindgen` to use the version of `libclang` installed via `brew install llvm` rather than the system one, I also need to `export LIBCLANG_PATH=/usr/local/opt/llvm/lib`.

With that done, `cargo clean && cargo espflash --release --monitor /dev/tty.usbmodem14101` worked like a charm and I'm now staring at a green and blue blinkenlight.

doxxx, 2022-09-15:
@davidcornu I've tried that and now its failing with a couple of rust errors:
* `cannot find macro 'llvm_asm' in this scope`
* `unknown feature 'llvm_asm'`.

doxxx, 2022-09-15:
I also get a bunch of warnings at the start:

```
[Thu 11:13am] gordon@Taiga ~/d/e/e/i/hardware-check> cargo build --release                                                                             (main|✚3…1)
info: syncing channel updates for 'nightly-2022-03-10-x86_64-apple-darwin'
info: latest update on 2022-03-10, rust version 1.61.0-nightly (458262b13 2022-03-09)
warning: Force-skipping unavailable component 'rust-std-riscv32imc-esp-espidf'
warning: patch for the non root package will be ignored, specify patch at the workspace root:
package:   /Users/gordon/dev/esp32-rust/espressif-trainings/common/lib/esp32-c3-dkc02-bsc/Cargo.toml
workspace: /Users/gordon/dev/esp32-rust/espressif-trainings/Cargo.toml
warning: profiles for the non root package will be ignored, specify profiles at the workspace root:
package:   /Users/gordon/dev/esp32-rust/espressif-trainings/intro/hardware-check/Cargo.toml
workspace: /Users/gordon/dev/esp32-rust/espressif-trainings/Cargo.toml
warning: patch for the non root package will be ignored, specify patch at the workspace root:
package:   /Users/gordon/dev/esp32-rust/espressif-trainings/intro/hardware-check/Cargo.toml
workspace: /Users/gordon/dev/esp32-rust/espressif-trainings/Cargo.toml
warning: resolver for the non root package will be ignored, specify resolver at the workspace root:
package:   /Users/gordon/dev/esp32-rust/espressif-trainings/intro/hardware-check/Cargo.toml
workspace: /Users/gordon/dev/esp32-rust/espressif-trainings/Cargo.toml
```

doxxx, 2022-09-15:
I pulled the latest commits for this repo and retried your suggestions, having cleaned up my unwise modifications, and now its working! Blue & green blinkenlitzen. :)

closed 2022-09-15 by doxxx