esp-rs/std-training #109: misleading troubleshooting instructions

spookyvision, 2022-09-15:
in [hello board](https://espressif-trainings.ferrous-systems.com/02_4_hello_board.html) it says (reverting the order for clarity)

> 1. press and hold boot button on the board, start flash command, release boot button after flashing process starts

this is incomplete and holds the user hostage for an unnecessarily long amount of time.
A little backstory: 
What is very likely the case here is that the [ESP Rust Board](https://github.com/esp-rs/esp-rust-board) - unlike the dev kit boards - lacks the required external components to [put the mcu in DFU/bootloader mode](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html) (I see no dedicated chip to assert `DTR` and `RTS`).
The solution is to do this manually by 

1. pressing and holding the boot button
2. pressing and releasing the reset button
3. releasing the boot button 
(aka you don't have to hold anything down while flashing)

furthermore (this is the "incomplete" part), unfortunately this means that `cargo espflash --monitor` will not ever work on these boards, since after flashing has completed you
4. have to reset the board once more to exit DFU mode and run the newly uploaded firmware
5. start `espmonitor` after said reset has been triggered.

> 2. use a hub.

I'm not aware of a situation where this would help. Did you maybe specifically mean "use a *powered* hub"? (although I'd say that if that helps your current hub isn't "unpowered" it's "running outside of spec")

Xylakant, 2022-09-15:
"use a hub" is correct - there are a few reports that using hubs fix download/flashing issues, for example https://github.com/espressif/esptool/issues/712#issuecomment-1002955244

spookyvision, 2022-09-15:
> "use a hub" is correct - there are a few reports that using hubs fix download/flashing issues, for example [espressif/esptool#712 (comment)](https://github.com/espressif/esptool/issues/712#issuecomment-1002955244)

ok, wow, so this seems to be a lot more cursed than I initially assumed, and one person even says a *powered* hub might interfere in a *negative* way (not sure I'm buying it in this instance, but eh, it's hardware, you never know). 

Also, since the flashing process does work for tanks my initial assumption about missing `DTR`/`RTS` can't be quite right. My next theory was that timing issues might occur when using a very fast (USB-C/USB-3) USB setup, so I've connected the filthiest oldest USB2 hub I could find, but that didn't fix it for me either. 

I'll close this issue and open a new one for clarified wording only.

closed 2022-09-15 by spookyvision