esp-rs/std-training #118: embedded-io compilation fails

JuliDi, 2022-10-27:
After following the instructions in the book, I tried building the Hello Board project with `cargo build --target riscv32imc-esp-espidf`.
The build then fails with many errors of the following type:
```
   Compiling embedded-io v0.3.1
error[E0658]: generic associated types are unstable
  --> /home/<user>/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:17:5
   |
17 | /     type ReadFuture<'a>: Future<Output = Result<usize, Self::Error>>
18 | |     where
19 | |         Self: 'a;
   | |_________________^
   |
   = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
   = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable
```

My rust toolchain is up to date and the specific version requested in the book is available:
```
$ rustup toolchain list
stable-x86_64-unknown-linux-gnu (default)
nightly-2022-03-10-x86_64-unknown-linux-gnu (override)
```

JakobGalaxy, 2022-10-27:
I am experiencing the same problem.

JakobGalaxy, 2022-10-27:
Ok, so I am still new to Rust and a hardly know what I am doing here, but it seems to work, so here it is:

After some searching I found [this Reddit post](https://www.reddit.com/r/rust/comments/ppsft4/generic_associated_types_are_unstable/), which says that the feature must be enabled using 
`#![feature(generic_associated_types)]` in order to work.

The error says the problem occurs in this file: `~/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs`, so I opened the `src` directory containing the `asynch.rs` and added the before-mentioned line 
`#![feature(generic_associated_types)]` to the top of the `lib.rs` file.

Now it works for me. I don't really think that this is a solution, but I also don't think that it can break much either, as you can always just delete the `~/.cargo/registry/` directory to revert this change. Hope this might help!

JuliDi, 2022-10-28:
Thanks for the hint! I was thinking about doing what the compiler suggests, as well. However, this can't be a permanent solution.
FYI you do not need to mess with the registry afaik, if you use this https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#testing-a-bugfix

So either embedded-io needs to fix this (which I doubt, because there v0.3.1 is very recent and fixes some issue with nightly), or this repo needs some change – but I have no clue which.

Moreover, it doesn't work on the docker container either. There is an issue with rustc being too old (1.61 and comfy-table needs 1.62 or newer).
Some crate in the dependency tree apparently causes a lot of trouble.

JakobGalaxy, 2022-10-28:
Thanks for the explanation!

JuliDi, 2022-10-28:
Just as an info for anyone else encountering the issue, the following steps can be used as a workaround:

- Clone the embedded-io repo from https://github.com/embassy-rs/embedded-io
- In `embedded-io/src/lib.rs` add the following code at the top of the file:
    ```rust
    #![feature(generic_associated_types)]
    ```
- In the training project's Cargo.toml (e.g. in `intro/hardware-check/Cargo.toml` add the following lines at the end
    ```toml
    [patch.crates-io]
    embedded-io = { path = "/path/to/embedded-io"}
    ```

fakeplant, 2022-11-28:
Easier temp solution than cloning repo. Just override to a prev package version.

```
[patch.crates-io]
embedded-io = { git = "https://github.com/embassy-rs/embedded-io.git", tag = "v0.3.0"}
```

and then `cargo update`

Thanks to @elotom
https://github.com/ivmarkov/rust-esp32-std-demo/issues/123#issuecomment-1300522982

JuliDi, 2022-11-28:
Thanks @fakeplant ! Works like a charm.
Embedded-io is now at release 0.4.0 but the issue seems to persist.

SergioGasquez, 2022-12-01:
This should now be fixed with #125

closed 2023-02-24 by SergioGasquez