esp-rs/std-training #122: CI build currently fails

Mirabellensaft, 2022-11-21:
CI build currently fails because a dependency of espflash 0.1.7 requires a newer nightly

BorealFeast, 2022-11-24:
error from building the .devcontainer/Dockerfile:
```
#7 186.6 error: failed to compile `cargo-espflash v1.7.0`, intermediate artifacts can be found at `/tmp/cargo-install9L4SSX`
#7 186.6
#7 186.6 Caused by:
#7 186.6   package `comfy-table v6.1.3` cannot be built because it requires rustc 1.62 or newer, while the currently active rustc version is 1.61.0-nightly
#7 186.6      Summary Successfully installed ldproxy! Failed to install cargo-espflash (see error(s) above).
#7 186.6 error: some crates failed to install
------
executor failed running [/bin/sh -c $HOME/.cargo/bin/cargo install cargo-espflash ldproxy]: exit code: 101
```

SergioGasquez, 2022-12-01:
With #125 merged this issue should be now fixed!