esp-rs/std-training #126: EspHttpServer.handle_get not found

IricBing, 2022-12-13:
[image]

EspHttpServer.handle_get methods not found at EspHttpServer struct when i test http server demo.

build error:

```
   Compiling http-server v0.1.0 (/home/iric/code/learn/espressif-trainings/intro/http-server)
error: could not compile `http-server` due to 3 previous errors; 2 warnings emitted
warning: unused imports: `io::Write`, `registry::Registry`
  --> examples/http_server.rs:10:20
   |
10 |     http::server::{registry::Registry, Response},
   |                    ^^^^^^^^^^^^^^^^^^
11 |     io::Write,
   |     ^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

error[E0599]: no method named `handle_get` found for struct `EspHttpServer` in the current scope
  --> examples/http_server.rs:31:12
   |
31 |     server.handle_get("/", |_request, response| {
   |            ^^^^^^^^^^ method not found in `EspHttpServer`

error[E0599]: no method named `handle_get` found for struct `EspHttpServer` in the current scope
  --> examples/http_server.rs:41:12
   |
41 |     server.handle_get("/temperature", move |_request, response| {
   |            ^^^^^^^^^^ method not found in `EspHttpServer`

warning: unused import: `Response`
  --> examples/http_server.rs:10:40
   |
10 |     http::server::{registry::Registry, Response},
   |                                        ^^^^^^^^

error: aborting due to 2 previous errors; 2 warnings emitted

```

m1racoli, 2022-12-23:
Same issue here. After some digging i found the previous implementation of the example, which indeed works. Take a look at the change.

https://github.com/ferrous-systems/espressif-trainings/commit/521ba1827f1d172ec59393091bbb1b7434e751ba

This worked for me after all.

```rust
use embedded_svc::{
    http::{
        server::{registry::Registry,Response, ResponseWrite},
        Method,
    },
    io::Write,
};

fn main() -> anyhow::Result<()> {
    # ...
    server.set_inline_handler("/", Method::Get, |request, response| {
        let html = index_html();
        let mut writer = response.into_writer(request)?;
        writer.do_write_all(html.as_bytes())?;
        writer.complete()
    })?;
```

m1racoli, 2022-12-24:
I have the suspicion that the `http` module has not only changed in later versions, but also is now only available behind the `experimental` flag. The docs do not show the module, but it is available in the source code.
https://docs.rs/embedded-svc/latest/src/embedded_svc/lib.rs.html#21

closed 2023-03-01 by SergioGasquez