esp-rs/std-training #127: executing command 'cargo espflash'

luigizuccarelli, 2023-01-08:
I recently purchased a couple of ESP32-Rust dev boards to check out rust on the esp32.

After following the instructions to install and setup rust and the online workbook, I get the following error when executing the command in the ```intro/hardware-check``` directrory

```
cargo espflash --release --monitor /dev/ttyACM0
New version of cargo-espflash is available: v2.0.0-rc.2

Serial port: /dev/ttyACM0
Connecting...

   Compiling serde v1.0.152
   Compiling thiserror v1.0.38
   Compiling ignore v0.4.19
   Compiling strum v0.24.1
   Compiling proc_macro v0.0.0 (/home/lzuccarelli/.rustup/toolchains/nightly-2022-03-10-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/proc_macro)
   Compiling darling_macro v0.14.2
error: could not compile `ignore` due to 4 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
error[E0658]: use of unstable library feature 'scoped_threads'
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1285:9
     |
1285 |         std::thread::scope(|s| {
     |         ^^^^^^^^^^^^^^^^^^
     |
     = note: see issue #93203 <https://github.com/rust-lang/rust/issues/93203> for more information
     = help: add `#![feature(scoped_threads)]` to the crate attributes to enable

error[E0658]: use of unstable library feature 'scoped_threads'
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1299:32
     |
1299 |                 handles.push(s.spawn(|| worker.run()));
     |                                ^^^^^
     |
     = note: see issue #93203 <https://github.com/rust-lang/rust/issues/93203> for more information
     = help: add `#![feature(scoped_threads)]` to the crate attributes to enable

error[E0658]: use of unstable library feature 'scoped_threads'
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1302:24
     |
1302 |                 handle.join().unwrap();
     |                        ^^^^
     |
     = note: see issue #93203 <https://github.com/rust-lang/rust/issues/93203> for more information
     = help: add `#![feature(scoped_threads)]` to the crate attributes to enable

error: aborting due to 3 previous errors
```

I tried adding the ```#![feature(scoped_threads)]``` to the lib.rs, wifi.rs and both build.rs and src/main.rs in the intro/hardware-check directory.

The rustc version is

```
rustc --version
rustc 1.61.0-nightly (458262b13 2022-03-09)

```

Any assistance on how to fix this will be appreciated

TAMHAN, 2023-01-15:
Hello,
I see exactly the same error on Ubuntu 20.04LTS

luigizuccarelli, 2023-01-15:
I added ```#![feature(scoped_threads)]``` to lib.rs in the ignore-0.4.19/src directory. The error I'm getting is this now

```
rror: could not compile `ignore` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
error[E0495]: cannot infer an appropriate lifetime due to conflicting requirements
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1289:30
     |
1289 |                     visitor: builder.build(),
     |                              ^^^^^^^^^^^^^^^
     |
note: first, the lifetime cannot outlive the anonymous lifetime #2 defined here...
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1229:5
     |
1229 |     pub fn visit(mut self, builder: &mut dyn ParallelVisitorBuilder<'_>) {
     |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: ...so that the declared lifetime parameter bounds are satisfied
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1289:30
     |
1289 |                     visitor: builder.build(),
     |                              ^^^^^^^^^^^^^^^
note: but, the lifetime must be valid for the anonymous lifetime #1 defined here...
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1285:28
     |
1285 |           std::thread::scope(|s| {
     |  ____________________________^
1286 | |             let mut handles = vec![];
1287 | |             for _ in 0..threads {
1288 | |                 let worker = Worker {
...    |
1303 | |             }
1304 | |         });
     | |_________^
note: ...so that the types are compatible
    --> /home/lzuccarelli/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.19/src/walk.rs:1299:32
     |
1299 |                 handles.push(s.spawn(|| worker.run()));
     |                                ^^^^^
     = note: expected `&Scope<'_, '_>`
                found `&Scope<'_, '_>```

It seems like ignore has undergone some changes from the initial 0.4.11 that was in the initial Carg.toml for globwalk-o.8.1

ashnikel, 2023-01-17:
Hi!
Add `ignore = "=0.4.11"` to `[dependencies]` section in Cargo.toml and then run `cargo clean`. That should do the trick.

SergioGasquez, 2023-01-20:
Just created #128 adding the dependency to all the required places for all the examples to build! Thanks @ashnikel for the fix suggestion!

luigizuccarelli, 2023-01-21:
@SergioGasquez - I pulled your PR locally it seems like the ignore issue is fixed . 
However I'm getting this error now 

```
gcc -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/python3.11 -I/usr/include/python3.11 -I/tmp/pip-install-i4hhnq2r/gevent_2d7929b8b5f541aa9a4198efec9c0075/deps -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -I/home/lzuccarelli/.espressif/python_env/idf4.4_py3.11_env/include -I/usr/include/python3.11 -c src/gevent/_greenlet_primitives.c -o build/temp.linux-x86_64-cpython-311/src/gevent/_greenlet_primitives.o
        src/gevent/_greenlet_primitives.c:216:12: fatal error: longintrepr.h: No such file or directory
          216 |   #include "longintrepr.h"
              |            ^~~~~~~~~~~~~~~
        compilation terminated.
        error: command '/usr/bin/gcc' failed with exit code 1
        [end of output]
    
    note: This error originates from a subprocess, and is likely not a problem with pip.
    ERROR: Failed building wheel for gevent
  ERROR: Could not build wheels for gevent, which is required to install pyproject.toml-based projects
  Traceback (most recent call last):
    File "/home/lzuccarelli/.espressif/esp-idf/v4.4.1/tools/idf_tools.py", line 1858, in <module>
      main(sys.argv[1:])
    File "/home/lzuccarelli/.espressif/esp-idf/v4.4.1/tools/idf_tools.py", line 1854, in main
      action_func(args)
    File "/home/lzuccarelli/.espressif/esp-idf/v4.4.1/tools/idf_tools.py", line 1575, in action_install_python_env
      subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr, env=env_copy)
    File "/usr/lib64/python3.11/subprocess.py", line 413, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['/home/lzuccarelli/.espressif/python_env/idf4.4_py3.11_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/lzuccarelli/.espressif/esp-idf/v4.4.1/requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.
  Error: Could not install esp-idf

  Caused by:
      command '"python3" "/home/lzuccarelli/.espressif/esp-idf/v4.4.1/tools/idf_tools.py" "--idf-path" "/home/lzuccarelli/.espressif/esp-idf/v4.4.1" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1
```

This step is now failing `error: failed to run custom build command for 'esp-idf-sys v0.31.5'`
I cleaned the ` ~/..espressif/ `  dir to make sure previous installs were deleted 

Also looking around on StackOverflow (I seem to recall this problem before when installing esp-idf).

Any ideas ?

SergioGasquez, 2023-01-21:
> Any ideas?

This looks like something with your environment. Do you have python installed? If so, which version? If you are using `3.11`, try switching to `3.10`. Also, please, check if you have a `.embuild` directory under the project folder, if so, please delete it, `cargo clean`, and try to rebuild it.

Using the [Dockerfile](https://github.com/ferrous-systems/espressif-trainings/blob/main/.devcontainer/Dockerfile) I was able to [build all the projects successfully](https://github.com/ferrous-systems/espressif-trainings/actions/runs/3967309000/jobs/6799115387).

luigizuccarelli, 2023-01-22:
Cool thanks for the info on docker. I was using python 3.12~a3 (needed for some other project). 
I built the docker file and managed to get the code to compile successfully using `cargo build --release ` in the container

The document mentions using the host to flash. I tried with `cargo espflash flash ...`  but it  builds first , do you have an example command I can use that ignores the build (as it's already done in the container) that just flashes the binary ?

SergioGasquez, 2023-01-22:
> @SergioGasquez  Cool thanks for the info on docker. I was using python 3.12~a3 (needed for some other project). 
> I built the docker image and managed to get the code to compile successfully using `cargo build --release ` in the container
> 
> The document mentions using the host to flash. I tried with `cargo espflash flash ...`  but it  builds first , do you have an example command I can use that ignores the build/compile  (as it's already done in the container) that just flashes the binary ?

`espflash` is what you are looking for https://github.com/esp-rs/espflash/tree/main/espflash#usage

SergioGasquez, 2023-01-22:
> @SergioGasquez Cool thanks for the info on docker. I was using python 3.12~a3 (needed for some other project). I built the docker image and managed to get the code to compile successfully using `cargo build --release ` in the container

I would say that the python version was the issue, using a version between 3.6 and 3.10 should work.

PR #128 (2023-02-22): Add `ignore` dependency
As seen in #127, some of the examples were not building. As [suggested per @ashnikel](https://github.com/ferrous-systems/espressif-trainings/issues/127#issuecomment-1385523705) adding this dependency solves the issue.

Closes #127
closed 2023-02-22 by SergioGasquez