esp-rs/std-training #129: Build failures even inside the Docker container

davidstelter, 2023-01-30:
I'm running into lots of build issues following along on my macos laptop, so I decided to try out the provided Docker container. My first attempted build of the http_client example in the container failed too, all I did was clone this repository and run `cd espressif-trainings/intro/http-client && cargo build --release --example http_client`

```
error[E0658]: use of unstable library feature 'scoped_threads'
    --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.20/src/walk.rs:1285:9
     |
1285 |         std::thread::scope(|s| {
     |         ^^^^^^^^^^^^^^^^^^
     |
     = note: see issue #93203 <https://github.com/rust-lang/rust/issues/93203> for more information
     = help: add `#![feature(scoped_threads)]` to the crate attributes to enable

error[E0658]: use of unstable library feature 'scoped_threads'
    --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.20/src/walk.rs:1299:32
     |
1299 |                 handles.push(s.spawn(|| worker.run()));
     |                                ^^^^^
     |
     = note: see issue #93203 <https://github.com/rust-lang/rust/issues/93203> for more information
     = help: add `#![feature(scoped_threads)]` to the crate attributes to enable

error[E0658]: use of unstable library feature 'scoped_threads'
    --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ignore-0.4.20/src/walk.rs:1302:24
     |
1302 |                 handle.join().unwrap();
     |                        ^^^^
     |
     = note: see issue #93203 <https://github.com/rust-lang/rust/issues/93203> for more information
     = help: add `#![feature(scoped_threads)]` to the crate attributes to enable
```

Narukara, 2023-02-05:
Same as #127. Try adding `ignore = "=0.4.11"` to `Cargo.toml` like #128 did.

davidstelter, 2023-02-06:
Thanks, that got me past one set of errors, but there are more that follow (warnings removed for brevity):
`cargo build --release --example http_client`
```
Compiling esp32-c3-dkc02-bsc v0.1.0 (/Users/stelterd/code/ferrous-systems/espressif-trainings/espressif-trainings/common/lib/esp32-c3-dkc02-bsc)

error[E0599]: no method named `submit` found for struct `EspHttpRequestWrite` in the current scope
  --> examples/http_client.rs:50:31
   |
50 |     let mut response = writer.submit()?;
   |                               ^^^^^^ method not found in `EspHttpRequestWrite<'_>`

error[E0191]: the value of the associated type `Error` (from trait `Errors`) must be specified
  --> examples/http_client.rs:62:35
   |
62 |                 if let Ok(size) = Read::read(&mut reader, &mut buf){
   |                                   ^^^^ help: specify the associated type: `Read<Error = Type>`

error[E0599]: no function or associated item named `read` found for trait object `dyn embedded_svc::io::Read` in the current scope
  --> examples/http_client.rs:62:41
   |
62 |                 if let Ok(size) = Read::read(&mut reader, &mut buf){
   |                                         ^^^^ function or associated item not found in `dyn embedded_svc::io::Read`

Some errors have detailed explanations: E0191, E0599.
For more information about an error, try `rustc --explain E0191`.
warning: `http-client` (example "http_client") generated 4 warnings
error: could not compile `http-client` due to 3 previous errors; 4 warnings emitted
```

Narukara, 2023-02-08:
It seems that the code in the example has some errors. I made some changes to it and it works for me now. Here is the modified code:

```rust
...

let response = write.into_response()?;

...

match status {
    200..=299 => {
        let mut buf = [0_u8; 256];
        let mut reader = response.reader();
        loop {
            match reader.do_read(&mut buf) {
                Ok(size) if size > 0 => {
                    total_size += size;
                    let response_text = str::from_utf8(&buf[..size])?;
                    println!("{}", response_text);
                }
                _ => break,
            }
        }
    }
    _ => anyhow::bail!("unexpected response code: {}", status),
}

...
```

I think it is necessary to submit a PR to fix this issue.

SergioGasquez, 2023-03-01:
All the code and examples should now build successfully with no changes. CI now covers both the starting code and the `examples` folder with the “solutions” to it.

closed 2023-03-01 by SergioGasquez