esp-rs/std-training #150: hardware-check build fails

JanWendler, 2023-03-07:
# Error Message
error[E0063]: missing fields `enable_so_linger` and `linger_timeout` in initializer of `httpd_config`
  --> C:\Users\jan.wendler\.cargo\registry\src\github.com-1ecc6299db9ec823\esp-idf-svc-0.41.4\src\http\server.rs:68:14
   |
68 |         Self(httpd_config_t {
   |              ^^^^^^^^^^^^^^ missing `enable_so_linger` and `linger_timeout`

error[E0063]: missing fields `enable_so_linger` and `linger_timeout` in initializer of `httpd_config`
   --> C:\Users\jan.wendler\.cargo\registry\src\github.com-1ecc6299db9ec823\esp-idf-svc-0.41.4\src\httpd.rs:394:9
    |
394 |         esp_idf_sys::httpd_config_t {
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^ missing `enable_so_linger` and `linger_timeout`

error[E0308]: mismatched types
  --> C:\Users\jan.wendler\.cargo\registry\src\github.com-1ecc6299db9ec823\esp-idf-svc-0.41.4\src\ping.rs:37:18
   |
37 |             tos: conf.tos,
   |                  ^^^^^^^^ expected `i32`, found `u8`

Some errors have detailed explanations: E0063, E0308.
For more information about an error, try `rustc --explain E0063`.
error: could not compile `esp-idf-svc` due to 3 previous errors

# Description
I tried to run the hardware check with the newest commit [72a04ff](https://github.com/esp-rs/espressif-trainings/commit/72a04ff9aab7270196a1bcdcf89c1a5891ec0a47)
However the build fails in the very end. When searching for this problem online I found an issue discussing this problem on the esp-idf-svc repo. [esp-idf-svc/issues/143](https://github.com/esp-rs/esp-idf-svc/issues/143) and [esp-idf-svc/issues/151](https://github.com/esp-rs/esp-idf-svc/issues/151). Following this I tried to change the ESP-IDF version in the .cargo file to 4.4.2 but that did not solve the problem and the output is still the same.
I also tried to change the version to 4.4.3, which is the ESP-IDF version I have installed for C programming, but the result is the same.

SergioGasquez, 2023-03-07:
Hi!

Just tried with a fresh clone of the repo on a macOS m1 and had no issue with the `hardware-check` example, also the Dockerfile is able to [build it successfully](https://github.com/esp-rs/espressif-trainings/actions/runs/4353789179/jobs/7608271055). Can you elaborate on the system that you are using to build the example?

JanWendler, 2023-03-08:
I used an older repo and pulled the new changes. I just tried it with a new repo as well with the same result.
I'm on a Windows 10 machine. I do have a local ESP-IDF installation with the relevant environment variables (IDF_PATH and so on) set. I ran `cargo clean` several times.
For Rust, i have added the nightly-2022-03-10 toolchain and I'm running the hardware check with `cargo run`.
I will try to disable the local ESP-IDF installation an report again.

SergioGasquez, 2023-03-08:
> I will try to disable the local ESP-IDF installation an report again.

Having ESP-IDF environment variables can interfere with [`esp-idf-sys`](https://github.com/esp-rs/esp-idf-sys) and cause build errors, please, let me know how it goes!

JanWendler, 2023-03-08:
# Error Message with no ESP-IDF environment variables
Compiling esp-idf-hal v0.38.1
   Compiling embedded-svc v0.21.3
error: failed to run custom build command for `esp-idf-sys v0.31.5`

Caused by:
  process didn't exit successfully: `r:\intro\hardware-check\target\debug\build\esp-idf-sys-94bc043aa2a6ecfd\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  Python 3.7.5
  pip 22.3.1 from C:\Users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages\pip (python 3.7)
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Processing c:\users\jan.wendler\.espressif\esp-idf\v4.4.1\tools\kconfig_new\esp-windows-curses
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): finished with status 'done'
  Requirement already satisfied: setuptools>=21 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 4)) (65.6.3)
  Requirement already satisfied: click>=7.0 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 8)) (8.1.3)
  Requirement already satisfied: pyserial>=3.3 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 9)) (3.5)
  Requirement already satisfied: future>=0.15.2 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 10)) (0.18.2)
  Requirement already satisfied: cryptography>=2.1.4 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 12)) (38.0.4)
  Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 18)) (2.3.1)
  Requirement already satisfied: pyelftools>=0.22 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 19)) (0.29)
  Requirement already satisfied: idf-component-manager~=1.0 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 20)) (1.2.1)
  Requirement already satisfied: gdbgui==0.13.2.0 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 22)) (0.13.2.0)
  Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 25)) (0.9.0.2)
  Requirement already satisfied: python-socketio<5 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 28)) (4.6.1)
  Requirement already satisfied: jinja2<3.1 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 29)) (3.0.3)
  Requirement already satisfied: itsdangerous<2.1 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 30)) (2.0.1)
  Requirement already satisfied: kconfiglib==13.7.1 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 32)) (13.7.1)
  Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 35)) (1.5.4)
  Requirement already satisfied: bitstring>=3.1.6 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 36)) (4.0.1)
  Requirement already satisfied: ecdsa>=0.16.0 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 37)) (0.18.0)
  Requirement already satisfied: construct==2.10.54 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from -r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 41)) (2.10.54)
  Requirement already satisfied: Pygments<3.0,>=2.2.0 in c:\users\jan.wendler\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\requirements.txt (line 22)) (2.13.0)
[... cut ...]
    File "C:\Users\jan.wendler\.espressif\esp-idf\v4.4.1\tools\idf_tools.py", line 1098, in get_user_defined_targets
      if env == idf_env_json['idfSelectedId']:
  KeyError: 'idfSelectedId'
  Error: Could not install esp-idf

  Caused by:
      0: `idf_tools.py export` result contains no `IDF_PYTHON_ENV_PATH` item and no such environment variable found
      1: environment variable not found
warning: build failed, waiting for other jobs to finish...
error: build failed

# Description
So, I disabled all the environment variables and restarted my pc. I ran `cargo clean` and `cargo run`. Now I get this error.

JanWendler, 2023-03-08:
I found this issue [esp-idf/issues/9837](https://github.com/espressif/esp-idf/issues/9837) which recommended deleting ~/.espressif I did that and it worked! The project builds now.
Of course I ran into a new problem.

# Error mesage
     Running `espflash flash --release --monitor target\riscv32imc-esp-espidf\debug\hardware-check`
New version of espflash is available: v2.0.0-rc.3

error: Found argument '--release' which wasn't expected, or isn't valid in this context

        If you tried to supply `--release` as a value rather than a flag, use `-- --release`

USAGE:
    espflash.exe <SERIAL>

For more information try --help
error: process didn't exit successfully: `espflash flash --release --monitor target\riscv32imc-esp-espidf\debug\hardware-check` (exit code: 2)

# Description
I'm a bit confused why it says that a new version of espflash is available.  On [crates.io](https://crates.io/crates/espflash) v1.7.0 is the newest version.
I tried to find the error but this time I could not find a relevant issue. Any Idea what I'm doing wrong here?

SergioGasquez, 2023-03-08:
> # Description
> I'm a bit confused why it says that a new version of espflash is available. On [crates.io](https://crates.io/crates/espflash) > v1.7.0 is the newest version. I tried to find the error but this time I could not find a relevant issue. Any Idea what I'm doing wrong here?

`espflash v2.0.0-rc.3` is a release candidate, official release should land soon, you can install it with: `cargo install espflash@2.0.0-rc.3 --force` or you can [edit the custom runner](https://github.com/esp-rs/espressif-trainings/blob/main/intro/hardware-check/.cargo/config.toml#LL6C2-L6C2) to ` runner = "espflash  --monitor"`

JanWendler, 2023-03-08:
YES! I had to use the command `cargo install espflash --version 2.0.0-rc.3 --force` but now it works!

SergioGasquez, 2023-03-08:
>  is configured to use an older nightly version of Rust, which could not build the new espflash crate.

I am currently working on updating the whole training to use a more recent version of the nightly toolchain and of the crates that are used, you can see the changes here: https://github.com/SergioGasquez/espressif-trainings/tree/feature/crates-update

I'll close the issue, thanks for your reports!

closed 2023-03-08 by SergioGasquez