esp-rs/std-training #159: Python3.11 on Mac compile issues for `esp-idf-sys vX.X.X`

NickAger, 2023-03-20:
I found that I had to "degrade" my Python version to 3.9. I'd installed Python using homebrew so downgraded using:

```
brew unlink python@3.11
```

Might be worthwhile adding a note about Python 3.11 compatibility issues in: book/src/02_2_software.md

igrr, 2023-03-20:
Do you recall, what was the issue you have observed with Python 3.11?

MabezDev, 2023-03-20:
> I found that I had to "downgrade" my Python version to 3.9

Hmm, that's odd. I'm using v3.10 at the moment with no issues, did you try that or did you just go straight to 3.9? It would be great to see what the 3.11 error was if you can reproduce it.

NickAger, 2023-03-20:
@MabezDev I didn't try with Python v3.10

Here's a transcript:

```
brew list | grep python
python@3.10
python@3.11
python@3.7
python@3.8
python@3.9

% brew link python@3.11
==> Downloading https://formulae.brew.sh/api/formula.jws.json
######################################################################## 100.0%
Linking /usr/local/Cellar/python@3.11/3.11.2... 25 symlinks created.

% python3 --version
Python 3.11.2

% cargo clean

% cargo run
```
Compiling strum v0.23.0
   Compiling num_enum v0.5.11
   Compiling bytemuck v1.13.1
   Compiling rgb v0.8.36
   Compiling futures v0.3.27
   Compiling no-std-net v0.5.0
   Compiling enumset v1.0.12
   Compiling embedded-svc v0.24.0
    Building [======================>  ] 222/234: esp-idf-sys(build) 
error: failed to run custom build command for `esp-idf-sys v0.32.1`

Caused by:
  process didn't exit successfully: `/Users/nickager/programming/espressif-trainings/intro/hardware-check/target/debug/build/esp-idf-sys-c6afcc95ae75201b/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  PATH=/Users/nickager/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/nickager/.espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/nickager/.espressif/tools/ninja/1.10.2/:/Users/nickager/.espressif/esp-idf/v4.4.1/tools:$PATH
  pip 22.3.1 from /usr/local/lib/python3.11/site-packages/pip (python 3.11)
  created virtual environment CPython3.11.2.final.0-64 in 7777ms
    creator CPython3Posix(dest=/Users/nickager/.espressif/python_env/idf4.4_py3.11_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==23.0.1, setuptools==67.4.0, wheel==0.38.4
    activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring None: markers 'sys_platform == "win32"' don't match your environment
  Requirement already satisfied: setuptools>=21 in /Users/nickager/.espressif/python_env/idf4.4_py3.11_env/lib/python3.11/site-packages (from -r /Users/nickager/.espressif/esp-idf/v4.4.1/requirements.txt (line 4)) (67.4.0)
  Collecting click>=7.0
    Using cached https://dl.espressif.com/pypi/click/click-8.1.3-py3-none-any.whl (96 kB)
  Collecting pyserial>=3.3
    Using cached https://dl.espressif.com/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
  Collecting future>=0.15.2
    Using cached https://dl.espressif.com/pypi/future/future-0.18.3-py3-none-any.whl (492 kB)
  Collecting cryptography>=2.1.4
    Using cached https://dl.espressif.com/pypi/cryptography/cryptography-39.0.2-cp36-abi3-macosx_10_12_x86_64.whl (2.9 MB)
  Collecting pyparsing<2.4.0,>=2.0.3
    Using cached https://dl.espressif.com/pypi/pyparsing/pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
  Collecting pyelftools>=0.22
[... 103 lines cut ...]
          [
              "/Users/nickager/programming/espressif-trainings/intro/hardware-check/./sdkconfig.defaults",
          ],
      ),
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v4.4.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Tag("v4.4.1") }
    error: subprocess-exited-with-error
    
    × Building wheel for gevent (pyproject.toml) did not run successfully.
    │ exit code: 1
```

MabezDev, 2023-03-21:
> Failed to build gevent

Looks like we might have missing binary wheels for the `gevent` package with python 3.11. Will try and figure out how to solve this.

MabezDev, 2023-03-21:
Seems like the issues for 3.11 python wheels are still present (https://github.com/espressif/esp-idf/issues/10116). I imagine these will be solved over time, but for now, I recommend you stay on python 3.9.

closed 2023-03-21 by MabezDev