esp-rs/std-training #171: `cargo install cargo-generate` fails in dev containers

iot49, 2023-03-31:
Error: failed to compile `cargo-generate v0.18.1`, intermediate artifacts can be found at `/tmp/cargo-install140bht`

Fix: add `libssl-dev` (to apt-get) in Dockerfile.

iot49, 2023-03-31:
> Hi!
> The devcontainer already has cargo-generate installed
> Is there any reason why you want to compile it from source?

odd, I got an error that it was not installed. Must have done something
wrong.
Presumably you can close the issue.

SergioGasquez, 2023-03-31:
I just did a small test, and I was able to use `cargo-generate` from the devcontainer with no issue. I'll close the issue, should you have any problems using it, just let me know!

SergioGasquez, 2023-04-02:
As suggested, I will close the issue. Feel free to reopen it with reproduction steps if the issue still persists for you.