esp-rs/std-training #173: rust_analyzer::config failed to find any projects in [AbsPathBuf("/workspace")]

iot49, 2023-04-01:
I get the following errors from rust-analyzer (and the analyzer does not work):

```
[ERROR rust_analyzer::config] failed to find any projects in [AbsPathBuf("/workspace")]
[ERROR rust_analyzer::main_loop] FetchWorkspaceError:
rust-analyzer failed to discover workspace
```

I've also noticed that "remoteUser" changed in `devcontainer.json` from `root` to `esp`. With this change the container won't start.

SergioGasquez, 2023-04-02:
Do you mind sharing instructions on how to reproduce this? I was able to use the container with VsCode devcontainer and I also get RA working:
[image]

iot49, 2023-04-02:
* MacOS Ventura 13.1
* VSC Version: 1.77.0 (Universal), Electron: 19.1.11, Chromium: 102.0.5005.196
* Start with "open folder in container"
* rebuilding container created `esp` user, now logged in as `esp`
* `devcontainer.json` (exactly as on GitHub) gives the following Problems:
```
Property settings is not allowed. [Ln 9, Col 2]
Use 'customizations/vscode/extensions' instead
Property extensions is not allowed.
Use 'customizations/vscode/extensions' instead
```
* rust-analyzer v0.3.1451

[image]

* opening a .rs file (e.g. `hardware-check/src/main.rs`) results in rust-analyzer error:
```
[ERROR rust_analyzer::config] failed to find any projects in [AbsPathBuf("/workspace")]
[ERROR rust_analyzer::main_loop] FetchWorkspaceError:
rust-analyzer failed to discover workspace
```

[image]

* no run button, errors no flagged, etc.

iot49, 2023-04-02:
Repo:

```bash
esp@4e5d1c01ec6c:/workspace$ whoami
esp
esp@4e5d1c01ec6c:/workspace$ git remote -v
origin  https://github.com/esp-rs/espressif-trainings.git (fetch)
origin  https://github.com/esp-rs/espressif-trainings.git (push)
esp@4e5d1c01ec6c:/workspace$ git diff
diff --git a/.devcontainer/devcontainer.json b/.devcontainer/devcontainer.json
index 274b0c3..dd68912 100755
--- a/.devcontainer/devcontainer.json
+++ b/.devcontainer/devcontainer.json
@@ -17,7 +17,7 @@
     },
     "rust-analyzer.checkOnSave.command": "clippy",
     "[rust]": {
-      "editor.defaultFormatter": "rust-lang.rust-analyzer",
+      "editor.defaultFormatter": "rust-lang.rust-analyzer"
     }
   },
   "extensions": [
@@ -31,4 +31,4 @@
   "remoteUser": "esp",
   "workspaceMount": "source=${localWorkspaceFolder},target=/workspace,type=bind,consistency=cached",
   "workspaceFolder": "/workspace"
-}
+}
\ No newline at end of file
diff --git a/intro/hello-world/src/main.rs b/intro/hello-world/src/main.rs
index ad8a230..71b7807 100644
--- a/intro/hello-world/src/main.rs
+++ b/intro/hello-world/src/main.rs
@@ -12,3 +12,4 @@ fn main() {
         sleep(Duration::from_secs(3));
     }
 }
+
esp@4e5d1c01ec6c:/workspace$ 
```

iot49, 2023-04-02:
Update: I created an entirely new repo with just the hello-world app, and in this case the analyzer works. No idea why it does not with this cloned repo.

SergioGasquez, 2023-04-03:
Have you tried opening a project(e.g: intro/hardware-check) instead of the whole repo with VsCode?

iot49, 2023-04-04:
This works. But if I also want to run in a devcontainer I need to create a
separate one for each sub-directory (e.g. intro).

SergioGasquez, 2023-04-04:
> This works. But if I also want to run in a devcontainer I need to create a separate one for each sub-directory (e.g. intro).

What do you mean? You can open the devcontainer in the `espressif-trainings` repo, go to the dessired folder in the Vs Code terminal an run `code .`,  this will open the folder in VsCode using the same devcontainer

iot49, 2023-04-04:
Oh, great - I was not aware of this. Solved!

closed 2023-04-04 by iot49

FallProune1, 2023-12-27:
Al momento de crear un archivo en rust me aparece tambien ese error [ERROR rust_analyzer::config] failed to find any projects in [AbsPathBuf("//Desktop/Rust_de_0_a_heroe")] No se porque me aparece esto pero corro el archivo con esta extencion.
[image]
Por favor ayuda :v

SergioGasquez, 2023-12-27:
> Al momento de crear un archivo en rust me aparece tambien ese error 

Why are you truing to create a rust file? Mind-expanding on which project of the training are you trying to use? 

> No se porque me aparece esto pero corro el archivo con esta extencion.

I would suggest you use the usual cargo commands, at least, until you get it working.

And, please, let's try to keep the issue in English, so everyone can benefit from our conversation.

FallProune1, 2023-12-27:
Thanks, using cargo to create a Rust project worked.
Sorry, I'm a beginner in Rust, today I started learning the basics to get started.