esp-rs/std-training #180: Bump esp-idf tag version to v4.4.4 in hardware-check

Vollbrecht, 2023-05-05:
A user in the matrix chat had some problems, using the wifi. The problem he encountered was a wired network state his router pushed him in. This is happening when flashing the chip -> it starts and tries to connect to a wifi -> user than starts to attach with espmonitor -> reseting the esp (nuking the current connection)-> special network refuses to connect wifi again(looking user out for some minutes). There is a "fix" in the newer version of esp-idf that presents the user here with a more clear error.

SergioGasquez, 2023-05-08:
IMHO, if we bump version to `v4.4.4` it should be for all project, since it's a relevant change maybe we can discuss it in the next community meeting.

SergioGasquez, 2023-05-12:
> IMHO, if we bump version to `v4.4.4` it should be for all project, since it's a relevant change maybe we can discuss it in the next community meeting.

The issue was discussed in [yesterday's meeting ](https://github.com/esp-rs/rust/discussions/174) and we've decided to update the tag, I'll create a PR to handle it!

Vollbrecht, 2023-05-12:
> The issue was discussed in [yesterday's meeting ](https://github.com/esp-rs/rust/discussions/174) and we've decided to update the tag, I'll create a PR to handle it!

I could not made it yesterday. Thank you for addressing it.

closed 2023-05-12 by SergioGasquez