esp-rs/std-training #189: Flashing fails with FlashDeflateData

jmehrs, 2023-06-09:
Hey all,

I am trying to follow this `std-training` and believe that my board might be in a bad state as I am able to connect to it but I cannot flash it _fully_.
```
[00:00:00] [========================================]      13/13      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:00] [>                                       ]       1/405     0x10000
...
Error: espflash::timeout
  × Communication error while flashing device
  ╰─▶ Timeout while running FlashDeflateData command
```
Sometimes in the last stage of flashing I can get to 13/405 or 11/405 but it errors out regardless.

When attaching the `espflash monitor` to the board I see the following being repeated
[image]

Has anyone run into such an issue?
P.S.: Running on a Ubuntu 22.04 VM & using `espflash@2.0.0-rc3`

SergioGasquez, 2023-06-09:
Hi! This looks like a hardware issue to me, you could try using a different baud rate to flash (see ` -b, --baud <BAUD>` argument). Have you tried using another USB cable? Which devkit are you using?

Vollbrecht, 2023-06-09:
make sure that the 3 strapping pins used are in a stable non-floating configuration. for more info checkout the datasheet of the c3

SergioGasquez, 2023-06-09:
> make sure that the 3 strapping pins used are in a stable non-floating configuration. for more info checkout the datasheet of the c3

Yes, good point, here is the documentation for those pins in ESP32-C3: https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#select-bootloader-mode

MabezDev, 2023-06-19:
Please try `espflash@2.0.0-rc4`, `espflash@2.0.0-rc3` had a regression in flashing via usb-serial-jtag.

SergioGasquez, 2023-09-04:
Hi! I will be closing the issue for inactivity, please, feel free to reopen if the issue still persists

closed 2023-09-04 by SergioGasquez