esp-rs/std-training #198: Explanation about Active high/low

Narukara, 2023-07-08:
https://github.com/esp-rs/std-training/blob/main/book/src/05_reference.md?plain=1#L23C1-L29

I find this passage a bit confusing. I think there are three basic concepts here:

1. A digital signal has two states, 1 (high) and 0 (low).
2. Each signal can be active high or active low. It tells us which state can activate the function of this signal. For example, an active low RESET signal resets the chip only when it is logic low.
3. We usually use high voltage to represent logic 1 and low voltage to represent logic 0, which is called positive logic. There is also negative logic (low voltage for logic 1, high voltage for logic 0).

This passage seems to confuse these concepts. And I don't think LEDs are a good example (unless a schematic is provided), because [they can be driven in both ways](https://www.okdo.com/project/gpio-led/). The "active low pin" in the last paragraph is also a strange expression. Can we rephrase this passage?

SergioGasquez, 2023-07-26:
What about the following? 

> A digital signal can be high or low, represented by voltage differences. Positive logic uses high voltage for logic 1 and low voltage for logic 0. Negative logic reverses this. 
> Signals can be active high or active low, based on the logic state that activates them. 
> It is important to note that in embedded Rust, abstractions show the logic level and not the voltage level.

Narukara, 2023-07-27:
Sorry, I may not have expressed myself clearly. :cry: This question is a bit beyond my abilities. Maybe it's better to keep the status quo.

closed 2023-07-27 by Narukara