esp-rs/std-training #203: Is this conversion to UTF-8 string safe?

pghalliday, 2023-08-28:
https://github.com/esp-rs/std-training/blob/f7ce2e7d7afe520c027d836e81e1e50bd9d64caf/intro/http-client/examples/http_client.rs#L72C43-L72C43

It seems to me that there is a chance it will fail if the end of the buffer lands in the middle of a multibyte character. Or am I missing something?

i wondered if the choice of a 256 byte buffer would guarantee to keep things on character boundaries, but my understanding of UTF-8 suggests not.

Googling so far has not suggested a good solution to this (from_utf8_lossy doesn't seem to solve the issue). In my own solution I pushed everything into a Vector before doing the conversion, but that is obviously problematic. The next idea I had was to let it fail then try again with one less byte until it works, while remembering to pass the remainder back when I get another buffer full, but that would be quite complicated. So if we're going to go that route it might be better to think of a proper stream handling solution that consumes only complete characters until there are no more (probably in a ring buffer or something).

Anyway I just wanted to flag this in case it is an issue that has a known solution that I missed. As it is the error condition in the context of HTML is probably quite rare as the vast majority of mark up characters are single byte.

pghalliday, 2023-08-29:
Ok this implementation deals with invalid UTF-8 sequences too, but I'm not sure it's helpful in the context of the training materials.

```rust
const BUFFER_SIZE: usize = 256;

struct ResponsePrinter {
    // Fixed buffer to read into
    buffer: [u8; BUFFER_SIZE],
    // Offset into the buffer to indicate that there are still
    // bytes at the beginning that have not been decoded yet
    offset: usize,
}

impl ResponsePrinter {
    fn new() -> ResponsePrinter {
        ResponsePrinter {
            buffer: [0_u8; BUFFER_SIZE],
            offset: 0,
        }
    }

    fn print(&mut self, mut response: Response<&mut EspHttpConnection>) -> Result<()> {
        // Keep track of the total number of bytes read to print later
        let mut total = 0;
        loop {
            // read into the buffer starting at the offset to not overwrite
            // the incomplete UTF-8 sequence we put there earlier
            if let Ok(size) = response.read(&mut self.buffer[self.offset..]) {
                if size == 0 {
                    // no more bytes to read from the response
                    if self.offset > 0 {
                        bail!("Response ends with an invalid UTF-8 sequence with length: {}", self.offset)
                    }
                    break;
                }
                // Update the total number of bytes read
                total += size;
                // recursive print to handle invalid UTF-8 sequences
                self.print_utf8(size)?;
            }
        }
        println!("Total: {} bytes", total);
        Ok(())
    }

    fn print_utf8(&mut self, size: usize) -> Result<()> {
        // remember that we read into an offset and recalculate the
        // real length of the bytes to decode
        let size_plus_offset = size + self.offset;
        match str::from_utf8(&self.buffer[..size_plus_offset]) {
            Ok(text) => {
                // buffer contains fully valid UTF-8 data,
                // print it and reset the offset to 0
                print!("{}", text);
                self.offset = 0;
            },
            Err(error) => {
                // A UTF-8 decode error was encountered, print
                // the valid part and figure out what to do with the rest
                let valid_up_to = error.valid_up_to();
                print!("{}", str::from_utf8(&self.buffer[..valid_up_to])?);
                if let Some(error_len) = error.error_len() {
                    // buffer contains invalid UTF-8 data, print a replacement
                    // character then copy the remainder (probably valid) to the
                    // beginning of the buffer, reset the offset and deal with
                    // the remainder in a recursive call to print_utf8
                    print!("{}", char::REPLACEMENT_CHARACTER);
                    let valid_after = valid_up_to + error_len;
                    self.buffer.copy_within(valid_after.., 0);
                    self.offset = 0;
                    return self.print_utf8(size_plus_offset - valid_after);
                } else {
                    // buffer contains incomplete UTF-8 data, copy the invalid
                    // sequence to the beginning of the buffer and set an offset
                    // for the next read
                    self.buffer.copy_within(valid_up_to.., 0);
                    self.offset = size_plus_offset - valid_up_to;
                }
            }
        }
        Ok(())
    }
}
```

SergioGasquez, 2023-08-31:
Hi! Thanks for opening the issue and sharing your findings on the topic! Would you mind opening a PR with your solution? For the purpose of the training, I would keep it as simple as possible as the main point of the exercise is the HTTP request.

pghalliday, 2023-09-02:
I'll cut it down to make it more palatable but at least safe for the happy path of valid utf-8 :)

PR #204 (2023-09-04): handle the case where UTF-8 sequences are split by the chunking in the http-client examples
Closes #203
closed 2023-09-04 by SergioGasquez