esp-rs/std-training #213: Panic when trying to read temperature

GamerBene19, 2023-11-08:
Environment: 
- ESP32-C3-DevKitM-1
- connected via USB cable to an USB3 port on my computer

Every time I try to read a value from the internal temperature sensor (e.g. with the http-server example) the ESP panics with
```
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200083c  RA      : 0x420006f0  SP      : 0x3fc9ab80  GP      : 0x3fc90c00  
0x4200083c - http_server::main
    at ??:??
0x420006f0 - http_server::main
    at /path/to/std-training/intro/http-server/examples/http_server.rs:55
0x3fc9ab80 - _heap_start
    at ??:??
...
```
From my debugging it seems to occur when `start_measurement` is executed.

I tried googling for the issue and found [this issue](https://github.com/espressif/arduino-esp32/issues/2547) which suggests it might be power related. I have plugged in the ESP into ports that are able to supply higher current (powerbank and phone charger), but it seems to experience the same error.
Note: Since I don't have a serial interface on those ports I used the LED to indicate statuses and the color I programmed after the measurement never showed.

I've also tried the code from [here](https://www.espboards.dev/blog/esp32-inbuilt-temperature-sensor/) to read the temperature in the Arduino IDE, which **works**.

Any advice would be greatly appreciated, thanks in advance!

SergioGasquez, 2023-11-08:
The training is intended for [ESP32-C3-DevKit-RUST-1](https://github.com/esp-rs/esp-rust-board) which has an SHTC3 sensor, the ESP32-C3-DevKitM-1 does not contain such sensor, hence the error.

SergioGasquez, 2023-11-08:
I am working on adding support for Wokwi simulation to the exercises (#168 #154) so soon, you should be able to follow most of the training even without the hardware

GamerBene19, 2023-11-08:
> The training is intended for [ESP32-C3-DevKit-RUST-1](https://github.com/esp-rs/esp-rust-board) which has an SHTC3 sensor, the ESP32-C3-DevKitM-1 does not contain such sensor, hence the error.

That explains a lot. Thanks for pointing out - somehow I seem to have overlooked that. I just assumed this training would use the [ESP32C3's internal temperature sensor](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/temp_sensor.html) which it seems it does not.

You might be the wrong person, but do you know if there is a way to access the internal sensor (linked above) via Rust? [It seems to be possible in C](https://github.com/espressif/esp-idf/blob/b90dfe075923870a8d1f8d85a75318781455f57a/examples/peripherals/temperature_sensor/temp_sensor/main/temp_sensor_main.c) as linked in the example above. 
I've looked in https://github.com/esp-rs/esp-idf-hal and it does not seem like it contains a driver for the temperature sensor. But I'm not sure if it would be in that repository or if I would have to look somewhere else.

Perhaps you (or someone else) can answer some of my questions.

SergioGasquez, 2023-11-09:
Just saw that you opened https://github.com/esp-rs/esp-idf-hal/issues/333, I will be closing this issue in favor of that one.

closed 2023-11-09 by SergioGasquez

GamerBene19, 2023-11-09:
> Just saw that you opened [esp-rs/esp-idf-hal#333](https://github.com/esp-rs/esp-idf-hal/issues/333), I will be closing this issue in favor of that one.

Thanks, I forgot to close it yesterday evening.