esp-rs/std-training #230: ESP32-C6 and ZigBee examples

Utopiah, 2024-01-08:
Hi, this looks like an amazing resources.

It does mentions "It is possible to follow the intro part with [ESP32-C3-DevKitC-02](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html) but, we don't recommend it. It is inherently easier to follow the training when using the same hardware." but I'm wondering how pragmatic it would be to follow for the ESP32-C6-DevKitC-1?

If it is feasible, how would going from existing examples to specific examples, e.g Zigbee support, e.g becoming a device for Home Assistant, require?

SergioGasquez, 2024-01-15:
Following the training with C6 should be possible, although projects that require specific sensors presents on the Rust ESP Board won't be available, but the same happens for ESP32-C3-DevKitC-02.

> If it is feasible, how would going from existing examples to specific examples, e.g Zigbee support, e.g becoming a device for Home Assistant, require?

We definitely can add more example projects, but those should be able to run on the Rust ESP board, so examples like Zigbee are not really an option.

Utopiah, 2024-01-15:
Obviously yes sensors not present won't work but even running any code, like starting with a hello world from the serial console, that is Rust is already an exciting step IMHO.

Now I just ran https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_light few days ago. If I was to run the hello world example from esp-rs, where should I look for in order to get Zigbee support, even if not in examples (as I understanding it would be out of scope).

SergioGasquez, 2024-01-16:
> like starting with a hello world from the serial console, that is Rust is already an exciting step IMHO.

There are several simple examples in https://github.com/esp-rs/esp-idf-hal/tree/master/examples. For more complex projects, see https://github.com/esp-rs/awesome-esp-rust?tab=readme-ov-file#projects

> where should I look for in order to get Zigbee support

There is no support for Zigbee, yet. We are currently working on the PHY/MAC level stuff in https://github.com/esp-rs/esp-ieee802154/https://github.com/esp-rs/esp-openthread

Utopiah, 2024-01-16:
Ah! Wonderful I'm now watching https://github.com/esp-rs/esp-ieee802154 and https://github.com/esp-rs/esp-openthread to learn how this works.

Closing the issue and hoping others will be able to find out the potential to learn more about esp-rs and Zigbee/Thread through this conversation.

closed 2024-01-16 by Utopiah

toonsevrin, 2025-01-02:
Zigbee support would be awesome.