esp-rs/std-training #231: The MQTT lesson is not for ESP32?

hongquan, 2024-01-10:
I am following the [MQTT lesson](https://github.com/esp-rs/std-training/tree/main/intro/mqtt/host-client). When I add `rumqtt` as dependency, it doesn't compile, because it depends on `tokio` -> `mio`, which in turn requires Linux uring.
So, can your confirm that this lesson for Linux host, not for ESP32 device?

MabezDev, 2024-01-10:
Tokio has `espidf` support. Can you post your error?

SergioGasquez, 2024-01-10:
There are 2 projects for  the MQTT lesson:
- ESP32 project, under `intro/mqtt/exercise`
  - This is meant to run in your esp32
- Computer project, under `intro/mqtt/host-client`
  - This is meant to run in your computer

In the book:
> ✅ In intro/mqtt/host_client you can find a host run program that mimics the behavior of a second client. Run it in a separate terminal using the cargo run command. Find more information about the host client below.

closed 2024-01-11 by hongquan