esp-rs/std-training #232: Running "hardware-check" reports "This script was called from a virtual environment, can not create a virtual environment again" error

gudaoxuri, 2024-01-17:
## Environment

esp-idf : v5.1.2
Rust : 1.75.0
OS: Windows 11

## Steps

* Open ``ESP-IDF 5.1 CMD`` and go to the ``hardware-check`` directory
* Execute ``cargo build``

## Error message

```
   Compiling compiler_builtins v0.1.103
   Compiling proc-macro2 v1.0.76
   Compiling unicode-ident v1.0.12
   Compiling core v0.0.0 (C:\Users\i\.rustup\toolchains\nightly-2023-11-14-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core)
   Compiling libc v0.2.150
   Compiling std v0.0.0 (C:\Users\i\.rustup\toolchains\nightly-2023-11-14-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std)
   Compiling serde v1.0.195
   Compiling windows_x86_64_msvc v0.48.5
   Compiling windows_x86_64_msvc v0.52.0
   Compiling memchr v2.7.1
   Compiling winapi v0.3.9
   Compiling crossbeam-utils v0.8.19
   Compiling syn v1.0.109
   Compiling regex-syntax v0.8.2
   Compiling glob v0.3.1
   Compiling windows_x86_64_msvc v0.42.2
   Compiling cfg-if v1.0.0
   Compiling rustversion v1.0.14
   Compiling anyhow v1.0.75
   Compiling aho-corasick v1.1.2
   Compiling libc v0.2.152
   Compiling rustix v0.38.30
   Compiling clang-sys v1.7.0
   Compiling log v0.4.20
   Compiling windows-targets v0.52.0
   Compiling stable_deref_trait v1.2.0
   Compiling windows-targets v0.48.5
   Compiling windows-sys v0.52.0
   Compiling as-slice v0.2.1
   Compiling windows-sys v0.48.0
   Compiling bstr v1.9.0
   Compiling thiserror v1.0.56
   Compiling once_cell v1.19.0
   Compiling minimal-lexical v0.2.1
   Compiling bitflags v2.4.1
   Compiling quote v1.0.35
   Compiling crossbeam-epoch v0.9.18
   Compiling heck v0.4.1
   Compiling syn v2.0.48
   Compiling semver v1.0.21
   Compiling serde_json v1.0.111
   Compiling crossbeam-deque v0.8.5
   Compiling nom v7.1.3
   Compiling aligned v0.4.1
   Compiling windows-targets v0.42.2
   Compiling cvt v0.1.2
   Compiling regex-automata v0.4.3
   Compiling bitflags v1.3.2
   Compiling itoa v1.0.10
   Compiling bindgen v0.63.0
   Compiling ryu v1.0.16
   Compiling either v1.9.0
   Compiling lazy_static v1.4.0
   Compiling windows-sys v0.45.0
   Compiling peeking_take_while v0.1.2
   Compiling shlex v1.2.0
[... 136 lines cut ...]
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "C:\\Espressif\\frameworks\\esp-idf-v5.1.2",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using custom user-supplied esp-idf repository at 'C:\Espressif\frameworks\esp-idf-v5.1.2' (detected from env variable `IDF_PATH`)
  ERROR: This script was called from a virtual environment, can not create a virtual environment again
  Error: Could not install esp-idf

  Caused by:
      command '"python" "C:\\Espressif\\frameworks\\esp-idf-v5.1.2\\tools\\idf_tools.py" "--idf-path" "C:\\Espressif\\frameworks\\esp-idf-v5.1.2" "--non-interactive" "install-python-env"' exited with non-zero status code 1
```

But the “Hello World” program is fine ( https://esp-rs.github.io/book/writing-your-own-application/generate-project/esp-idf-template.html )

What can I do to solve this?

SergioGasquez, 2024-01-17:
Please, try following the instructions in https://esp-rs.github.io/std-training/02_2_software.html#rust-toolchain and don't use an activated ESP-IDF, unless you are willing to modify the `.cargo/config.toml` of the projects (See https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir)

gudaoxuri, 2024-01-17:
> Please, try following the instructions in https://esp-rs.github.io/std-training/02_2_software.html#rust-toolchain and don't use an activated ESP-IDF, unless you are willing to modify the [...]

It's working. Thanks.

closed 2024-01-17 by gudaoxuri