esp-rs/std-training #235: Building `esp-idf-sys` failed because of uninitialized submodules in esp-idf

rangerjo, 2024-03-06:
Building `esp-idf-sys` failed because of uninitialized submodules in esp-idf.

Issuing `git submodle update --init --recursive` in the esp-idf directory at `~/.espressif/esp-idf/v5.1.2/` solved the issue.

However, when following the instructions in the std-training book, this is not mentioned, as this should all be handled under the hood. 

Just letting you know!

Vollbrecht, 2024-03-06:
Usually that should not be needed. When the esp-idf's repo is pulled and github decided to have bad networking again, problems like this can arise. Still thanks that you mention a potential workaround, and give feedback ;) 

Happy Coding

SergioGasquez, 2024-03-06:
`esp-idf-sys` should take care of downloading `esp-idf` with all the submodules. But sometimes, when not using a reliable network, the `esp-idf-sys` cloning of the `esp-idf` repo might fail, which looks like its what happened in your case

rangerjo, 2024-03-06:
Ok, thank you for the responses, thats fine with me. Can be considered closed then!

closed 2024-03-06 by rangerjo