esp-rs/std-training #241: How to save and pass 'PinDriver' objects

xiaguangbo, 2024-03-15:
My basic idea is to encapsulate a `onewire struct` and pass a decimal pin number to drive the corresponding pin. But how do you preserve the objects returned by `PinDriver: : input_output_od`? I'm not gonna write this：
```
struch Onewire {
    pin : ???
};

...

let mut onewire = PinDriver::input_output_od(0) // Use numbers instead, and if not match a separate line for each peripheral.pins.gpiox.
```

now：
```
use esp_idf_hal::delay::FreeRtos;
use esp_idf_svc::hal::{
    delay::Delay,
    gpio::{PinDriver, Pull},
    peripherals::Peripherals,
};

fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello, world!");

    let peripherals = Peripherals::take().unwrap();

    let mut onewire = PinDriver::input_output_od(peripherals.pins.gpio0).unwrap();
    onewire.set_pull(Pull::Floating).unwrap(); // Use external pull-up

    let delay = Delay::new_default();

    loop {
        FreeRtos::delay_ms(1);

        // reset
        onewire.set_low().unwrap();
        delay.delay_us(600);
        onewire.set_high().unwrap();
        delay.delay_us(80);

        if onewire.is_low() {
            delay.delay_us(800);

            if onewire.is_high() {
                log::info!("onewire: reset ok");
            } else {
                log::warn!("onewire: reset err");
                continue;
            }
        } else {
            log::warn!("onewire: reset err");
            continue;
        }
    }
}

```

SergioGasquez, 2024-03-15:
Hi! I would suggest using the matrix channel (https://matrix.to/#/#esp-rs:matrix.org) for this kind of questions as they are not issues per se and not strictly related with the training. Also, more people could jump in to help you.

xiaguangbo, 2024-03-15:
> Hi! I would suggest using the matrix channel (https://matrix.to/#/#esp-rs:matrix.org) for this kind of questions as they are not issues per se and not strictly related with the training. Also, more people could jump in to help you.

ok