esp-rs/std-training #242: WIFI Example does not compile: 'static str is not convertible to heapless::string::String<32>

juliankrieger, 2024-03-18:
When running a project templated by `esp-idf-template` and adding `wifi.rs` from `common`, I can not successfully run `cargo build`. 

I must admit, I'm a bit confused on the use of heapless in the esp_idf_svf::wifi Component. I thought the whole point of using the `std` feature flag was to enable the use of std::String.

Error: 
```
error[E0277]: the trait bound `heapless::string::String<64>: From<&str>` is not satisfied
  --> src/wifi.rs:59:24
   |
59 |         password: pass.into(),
   |                        ^^^^ the trait `From<&str>` is not implemented for `heapless::string::String<64>`, which is required by `&str: Into<_>`
   |
   = note: required for `&str` to implement `Into<heapless::string::String<64>>`
   ```

Vollbrecht, 2024-03-18:
Yeah that is not ideal. The overall correct way would be to update the training to the latest esp-idf-svc release. Thanks for bringing up the issue.

juliankrieger, 2024-03-18:
@Vollbrecht I wouldn't mind helping out.

Vollbrecht, 2024-03-18:
Feel free to create a PR, it probably would be enough to update every occurrence of esp-idf-svc in the Cargo.toml and test if all solution examples still work. Besides the mention wifi problem the esp-idf-svc API that is used in the training should not have changed much.

juliankrieger, 2024-03-18:
On it. What's to determine is if using .try_into.unwrap() is sufficient. Alternative, one would need to implement error handling in the style of anyhow for conversion errors.

Vollbrecht, 2024-03-18:
I think the training is using anyhow, so an expect() with a panic massage would be ok here

juliankrieger, 2024-03-18:
Done. Works on my machine. https://github.com/esp-rs/std-training/pull/243