esp-rs/std-training #252: hardware-check: Segment 0 0x3c090020-0x3c09fff8 invalid: bad load address range

bes, 2024-05-03:
I am trying to run the `hardware-check` intro app on the below device, which I think is an `ESP32-C3-DevKitC-02`:

```shell
cargo espflash board-info
[2024-05-03T12:22:41Z INFO ] Detected 2 serial ports
[2024-05-03T12:22:41Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-03T12:22:41Z INFO ] Please select a port
[2024-05-03T12:22:42Z INFO ] Serial port: '/dev/cu.usbserial-0001'
[2024-05-03T12:22:42Z INFO ] Connecting...
[2024-05-03T12:22:43Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.1)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       a0:b7:65:4d:6f:7c
```
[image]
[image]

Then I run `CRATE_CC_NO_DEFAULTS=1 cargo run --release`

```shell
    Finished release [optimized] target(s) in 42.23s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/release/hardware-check`
[2024-05-03T12:11:56Z INFO ] Detected 2 serial ports
[2024-05-03T12:11:56Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-03T12:11:56Z INFO ] Please select a port
[2024-05-03T12:12:00Z INFO ] Serial port: '/dev/tty.usbserial-0001'
[2024-05-03T12:12:00Z INFO ] Connecting...
[2024-05-03T12:12:01Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.1)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       a0:b7:65:4d:6f:7c
App/part. size:    791,776/4,128,768 bytes, 19.18%
[2024-05-03T12:12:02Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2024-05-03T12:12:02Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:45] [========================================]     477/477     0x10000          [2024-05-03T12:12:49Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
I (31) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=0ffd8h ( 65496) load
E (106) esp_image: Segment 0 0x3c090020-0x3c09fff8 invalid: bad load address range
E (115) boot: Factory app partition is not bootable
E (120) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
I (60) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (60) boot: compile time Jun  7 2023 07:48:23
I (63) boot: Multicore bootloader
I (67) boot: chip revision: v3.1
I (71) boot.esp32: SPI Speed      : 40MHz
I (76) boot.esp32: SPI Mode       : DIO
I (80) boot.esp32: SPI Flash Size : 4MB
I (85) boot: Enabling RNG early entropy source...
I (90) boot: Partition Table:
I (94) boot: ## Label            Usage          Type ST Offset   Length
I (101) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (109) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (116) boot:  2 factory          factory app      00 00 00010000 003f0000
I (124) boot: End of partition table
I (128) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=0ffd8h ( 65496) load
E (136) esp_image: Segment 0 0x3c090020-0x3c09fff8 invalid: bad load address range
E (145) boot: Factory app partition is not bootable
E (150) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46
```

I've tried to find similar errors on the internet, and tried a bunch of different solutions, but nothing worked for me.

I am grateful for any tips.

bes, 2024-05-03:
I forgot to write that I am on mac.

SergioGasquez, 2024-05-03:
Your chip is an ESP32, not an ESP32-C3
```
Chip type:         esp32 (revision v3.1)
```

bes, 2024-05-03:
Thanks!

Does this triple need to change then?

```toml
[build]
target = "riscv32imc-esp-espidf"
```

I tried googling but couldn't find what I was supposed to use.

SergioGasquez, 2024-05-03:
Yes, ESP32 is Xtensa while ESP32-C3 is RISC-V, so you need to use our Xtensa Rust toolchain (https://docs.esp-rs.org/book/installation/riscv-and-xtensa.html) and you would require more changes in code, here is an outdate version of the training for an Xtensa target (ESP32-S3): https://github.com/SergioGasquez/espressif-trainings/tree/esp32s3.

> Does this triple need to change then?

Your target should be `target = "xtensa-esp32-espidf"`

closed 2024-05-03 by bes

crawforc3, 2024-09-24:
The ESP32-CAM is also showing as ESP32 and probably needs to use the Xtensa Rust toolchain