esp-rs/std-training #253: RGB LED is not flashing if esp-idf-svc is 0.48.*

elosev, 2024-05-05:
RGB NeoPixel LED in hardware-check is not flashing if the crate is upgraded to the  latest 0.48.* version.  I have commented out the rest of the code but LED flashing and narrowed down the issue to this only dependency. 

Verified on ESP32-C3-DevKitC-02.

Vollbrecht, 2024-05-05:
Step 1) The libs in `https://github.com/esp-rs/std-training/tree/main/common/lib`need updating to 0.48 to work properly. 
Step 2) There were some small API changes in 0.48 that needs to be updated in the libs
Step 3) The rmt driver ( thing that is used to drive the LED machinery)  in the 0.48 release has a small bug - you would currently need to run the git master of esp-idf-svc 

So you either need to perform all 3 steps or use 0.47 in your generated test-project.

elosev, 2024-05-08:
I've done all 3, the LED is still not working.

I rolled back to 0.47.  Now I am getting https://github.com/esp-rs/esp-idf-svc/issues/344 which was fixed on master, but it is not clear when. The master is on 0.48 now. it does not seem the fix was backported to 0.47.

elosev, 2024-05-09:
Update: works now after getting esp-idf-hal from master.

closed 2024-06-24 by elosev