esp-rs/std-training #262: Unable to cargo run on hardware-check

adrien3d, 2024-07-10:
Hello,

My setup is working well with esp-idf when I am developping in C, and I've followed every step in the guide for [Rust](https://docs.esp-rs.org/std-training/02_2_software.html).

But, when running `cargo run`, I have the following result:
```
    It fails with the following output:

      Change Dir: /Users/adrien/Dev/esp/projects/std-training/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c3e48119a5703967/out/build/CMakeFiles/CMakeTmp

      Run Build Command(s):/Users/adrien/.espressif/tools/ninja/1.11.1/ninja cmTC_d1cd7 && [1/2] Building C object CMakeFiles/cmTC_d1cd7.dir/testCCompiler.c.obj
      FAILED: CMakeFiles/cmTC_d1cd7.dir/testCCompiler.c.obj
      /Users/adrien/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei  -ffunction-sections -fdata-sections --target=riscv32imc_zicsr_zifencei-esp-espidf -o CMakeFiles/cmTC_d1cd7.dir/testCCompiler.c.obj -c /Users/adrien/Dev/esp/projects/std-training/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c3e48119a5703967/out/build/CMakeFiles/CMakeTmp/testCCompiler.c
      riscv32-esp-elf-gcc: error: unrecognized command-line option '--target=riscv32imc_zicsr_zifencei-esp-espidf'
      ninja: build stopped: subcommand failed.

    CMake will not be able to correctly generate this project.
  Call Stack (most recent call first):
    /Users/adrien/Dev/esp/esp-idf/tools/cmake/project.cmake:506 (__project)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library/std/src/panicking.rs:661:5
     1: core::panicking::panic_fmt
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library/core/src/panicking.rs:74:14
     2: cmake::fail
               at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5
     3: cmake::run
               at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1076:9
     4: cmake::Config::build
               at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:813:13
     5: build_script_build::native::cargo_driver::build
               at ./build/native/cargo_driver.rs:527:5
     6: build_script_build::native::build
               at ./build/native.rs:17:9
     7: build_script_build::main
               at ./build/build.rs:51:24
     8: core::ops::function::FnOnce::call_once
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
➜  hardware-check git:(main) clear

➜  hardware-check git:(main) cargo run
   Compiling esp-idf-sys v0.35.0
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2]; explicitly specify an MCU to resolve this ambiguity
warning: esp-idf-sys@0.35.0: Ignoring activated esp-idf environment: ESP_IDF_TOOLS_INSTALL_DIR != fromenv
warning: esp-idf-sys@0.35.0: Ignoring configuration setting `ESP_IDF_VERSION` (Tag v5.2): custom esp-idf repository detected via $IDF_PATH

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/Users/adrien/Dev/esp/projects/std-training/intro/hardware-check/target/debug/build/esp-idf-sys-bcf67bf22311802d/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
[... 127 lines cut ...]
      
      Run Build Command(s):/Users/adrien/.espressif/tools/ninja/1.11.1/ninja cmTC_23ce2 && [1/2] Building C object CMakeFiles/cmTC_23ce2.dir/testCCompiler.c.obj
      FAILED: CMakeFiles/cmTC_23ce2.dir/testCCompiler.c.obj 
      /Users/adrien/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei  -ffunction-sections -fdata-sections --target=riscv32imc_zicsr_zifencei-esp-espidf -o CMakeFiles/cmTC_23ce2.dir/testCCompiler.c.obj -c /Users/adrien/Dev/esp/projects/std-training/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c3e48119a5703967/out/build/CMakeFiles/CMakeTmp/testCCompiler.c
      riscv32-esp-elf-gcc: error: unrecognized command-line option '--target=riscv32imc_zicsr_zifencei-esp-espidf'
      ninja: build stopped: subcommand failed.
      
      

    

    CMake will not be able to correctly generate this project.
  Call Stack (most recent call first):
    /Users/adrien/Dev/esp/esp-idf/tools/cmake/project.cmake:506 (__project)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Any idea on how I can fix this ?

Vollbrecht, 2024-07-10:
You mention that you have experience with C IDF framework. Does your current environment have a IDF version activated?
Can you check that you don't provide a `IDF_PATH` and a `ESP_IDF_VERSION` env, and thouse not overriding the values used in our .config.toml. 

In the training we setup a cloned version of ESP_IDF inside the .espressif dir according to the version that is set via the ESP_IDF_VERSION env variable and also try to populate the correct versioned compiler for that ESP-IDF version according the recommendation. 

Furthermore can you go into `/Users/adrien/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc` and check the output of `./riscv32-esp-elf-gcc --version`

adrien3d, 2024-07-10:
Thanks for your quick answer.
I've tried to unset both variables, but still not successfull:
```
➜  hardware-check git:(main) unset ESP_IDF_VERSION
➜  hardware-check git:(main) unset IDF_PATH       
➜  hardware-check git:(main) cargo run
   Compiling esp-idf-sys v0.35.0
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/Users/adrien/Dev/esp/projects/std-training/intro/hardware-check/target/debug/build/esp-idf-sys-bcf67bf22311802d/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2]; explicitly specify an MCU to resolve this ambiguity
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out '24c60e243580c7868f4334a1ba3123481fe1aa48'
  Submodule path 'components/bt/controller/lib_esp32': checked out '43ecd22ec6b5b484709693ae8e86478a7f130f17'
  Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out 'b9a902c3551ef4a2032b6662a4cbb018125ddfda'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '76ed4114ee7d081435a3c65793b4c8eb1dfaf199'
  Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out '0af472c6f95858e638565dad19a79fe47de1fff2'
  Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out '70612d08d1ff94dfb5da99d6a68aa4032cd63276'
  Submodule path 'components/bt/esp_ble_mesh/lib/lib': checked out '4934ca903807dd74f7f808dadcd9a478e18fc6c3'
  Submodule path 'components/bt/host/nimble/nimble': checked out 'a2acac3e14bb7810783ac5089ccf8229a2dd2f70'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/esp_coex/lib': checked out '2717868407a01878a49d19794538b6da120597ae'
  Submodule path 'components/esp_phy/lib': checked out '792ba5917ee8191e7264143e69f9e6f8c1c0eacc'
  Submodule path 'components/esp_wifi/lib': checked out 'd9a4b032efee9b50914cf38a6c1f41049811fa1e'
  Submodule path 'components/heap/tlsf': checked out 'd2e28f872472ffc6a704faae65ddee1f24e2dfba'
  Submodule path 'components/json/cJSON': checked out '87d8f0961a01bf09bef98ff89bae9fdec42181ee'
  Submodule path 'components/lwip/lwip': checked out 'f79221431fa9042b3572d271d687de66da7560c4'
  Submodule path 'components/mbedtls/mbedtls': checked out '72aa687352a469044cbb946f3fdb261430e41ce1'
  Submodule path 'components/mqtt/esp-mqtt': checked out 'aa6f889fb4f6f743b3a550aa587713aabbdca1fc'
  Submodule path 'components/openthread/lib': checked out 'a0f6a77960b36ebe357cc4bee280034f8c7120f1'
  Submodule path 'components/openthread/openthread': checked out 'be7d36e4ff9cf7df6dfce54e58a31163c87b93f7'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'abc67a11c6db271bedbb9f58be85d6f4e2ea8389'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  Python 3.10.1
  pip 24.1.2 from /Users/adrien/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
  Requirement already satisfied: pip in /Users/adrien/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (24.1.2)
  Requirement already satisfied: setuptools in /Users/adrien/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (70.3.0)
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
  Requirement already satisfied: setuptools in /Users/adrien/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from -r /Users/adrien/.espressif/esp-idf/v5.2.2/tools/requirements/requirements.core.txt (line 3)) (70.3.0)
  Requirement already satisfied: packaging in /Users/adrien/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from -r /Users/adrien/.espressif/esp-idf/v5.2.2/tools/requirements/requirements.core.txt (line 4)) (24.1)
  Requirement already satisfied: click in /Users/adrien/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from -r /Users/adrien/.espressif/esp-idf/v5.2.2/tools/requirements/requirements.core.txt (line 7)) (8.1.7)
[... cut ...]
    /Users/adrien/.espressif/esp-idf/v5.2.2/tools/cmake/project.cmake:506 (__project)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

For the command you asked me to run:
```
➜  bin ./riscv32-esp-elf-gcc --version
riscv32-esp-elf-gcc (crosstool-NG esp-13.2.0_20230928) 13.2.0
```

Vollbrecht, 2024-07-10:
can you please try to add the following in the `.cargo/config.toml` file under the [env] section: `CRATE_CC_NO_DEFAULTS = "1"` 

Test only that change with `cargo clean` + `cargo build`, and please tell if you observe a difference here.

 If that was not fruitful can you also test if there are changes when you change the `rustflag` line in the same file into the following ` rustflags = [ "--cfg",  "espidf_time64"]` e.g removing the default-linker-library stuff

adrien3d, 2024-07-10:
Thank you very much, it is working fine with `CRATE_CC_NO_DEFAULTS = "1"`
Can you tell me the meaning of that parameter?

Vollbrecht, 2024-07-10:
We should have had it included when we last updated the training but forgot. This is a env variable that is used by the cc-rs crate. The cc-rs crate is the thing that gets called from our embuild build-system to run the xtensa/riscv-gcc compiler. Since you now that esp-idf is using cmake, we also actually are using the cmake crate to interface with it, and it internally is going through cc-rs to call the compiler. So we don't directly control the calling to the cc-rs crate. Since we dont control it directly, we do that  with that env var. We instruct it to only use the flags that we explicitly want to use and don't pick something up from the systems default. 

If you see in the esp-rs/esp-idf-template we provide it in the .cargo/config.toml by default.

Vollbrecht, 2024-07-10:
i added it to all .configs in #263 so hopefully that should not be a problem anymore.

closed 2024-10-18 by SergioGasquez