esp-rs/std-training #264: 有蓝牙BLE的教程代码吗？

reformc, 2024-08-13:
想使用蓝牙的功能，但是项目里只有wifi。能介绍下蓝牙如何使用吗？

SergioGasquez, 2024-08-13:
Hi, take a look at the https://github.com/esp-rs/esp-idf-svc/blob/master/examples/bt_gatt_server.rs example.

Please, try to keep issues in English and for questions I'd recommend using the [Matrix channel](https://matrix.to/#/#esp-rs:matrix.org).

closed 2024-08-13 by SergioGasquez

reformc, 2024-08-16:
Thank you. I'm trying a Ble client instead of a Ble server because I'm trying to make a Ble controller device. Is there any codes for the Ble client? For example, how to negotiate MTU and how to perform long write

reformc, 2024-08-16:
> Hi, take a look at the https://github.com/esp-rs/esp-idf-svc/blob/master/examples/bt_gatt_server.rs example.
> 
> Please, try to keep issues in English and for questions I'd recommend using the [Matrix channel](https://matrix.to/#/#esp-rs:matrix.org).

Thank you. I'm trying a Ble client instead of a Ble server because I'm trying to make a Ble controller device. Is there any codes for the Ble client? For example, how to negotiate MTU, perform long write, notify