esp-rs/std-training #265: Instruction in the book for toolchain set-up on Debian/Ubuntu fails

nsengupta, 2024-10-15:
An Embedded Rust beginner here! So, please bear with me if the issues I raise are very fundamental and can be easily resolved or is already resolved! :-) 

From the page on the book [here](https://docs.esp-rs.org/std-training/02_2_software.html):

The instruction is to use the following command for installing the toolchain on Debian/Ubuntu:

`sudo apt install llvm-dev libclang-dev clang libuv-dev libuv1-dev pkgconf python3-venv python-is-python3
`

 On my Ubuntu 22.10 machine, this is what I get:

```
sudo apt install llvm-dev libclang-dev clang libuv-dev libuv1-dev pkgconf python3-venv python-is-python3
[sudo] password for nirmalya: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libuv-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libuv1-dev:i386 libuv1-dev

E: Package 'libuv-dev' has no installation candidate
```

Here is the output of `uname -a`:

`Linux bluewhale 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

I have removed the _libuv-dev_ from the command, and then the installation goes through:
```

sudo apt install llvm-dev libclang-dev clang libuv1-dev pkgconf python3-venv python-is-python3
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  binfmt-support clang-14 libclang-14-dev libclang-common-14-dev libclang-cpp14 libclang1-14 libpkgconf3 llvm llvm-14 llvm-14-dev llvm-14-linker-tools llvm-14-runtime llvm-14-tools
  llvm-runtime python3-pip-whl python3-setuptools-whl python3.10-venv
Suggested packages:
  clang-14-doc llvm-14-doc
The following packages will be REMOVED:
  pkg-config
The following NEW packages will be installed:
  binfmt-support clang clang-14 libclang-14-dev libclang-common-14-dev libclang-cpp14 libclang-dev libclang1-14 libpkgconf3 libuv1-dev llvm llvm-14 llvm-14-dev llvm-14-linker-tools
  llvm-14-runtime llvm-14-tools llvm-dev llvm-runtime pkgconf python-is-python3 python3-pip-whl python3-setuptools-whl python3-venv python3.10-venv
0 upgraded, 24 newly installed, 1 to remove and 30 not upgraded.
Need to get 106 MB of archives.
After this operation, 750 MB of additional disk space will be used.
Do you want to continue? [Y/n] y      
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libpkgconf3 amd64 1.8.0-1 [30.3 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/universe amd64 pkgconf amd64 1.8.0-1 [35.3 kB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/main amd64 binfmt-support amd64 2.2.1-2 [55.8 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 libclang-cpp14 amd64 1:14.0.0-1ubuntu1.1 [12.1 MB]
Get:5 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 libclang-common-14-dev amd64 1:14.0.0-1ubuntu1.1 [5,975 kB]
Get:6 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 llvm-14-linker-tools amd64 1:14.0.0-1ubuntu1.1 [1,355 kB]                                                                
Get:7 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 libclang1-14 amd64 1:14.0.0-1ubuntu1.1 [6,792 kB]                                                                        
Get:8 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 clang-14 amd64 1:14.0.0-1ubuntu1.1 [81.2 kB]                                                                             
Get:9 http://archive.ubuntu.com/ubuntu jammy/universe amd64 clang amd64 1:14.0-55~exp2 [3,558 B]                                                                                             
Get:10 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 libclang-14-dev amd64 1:14.0.0-1ubuntu1.1 [25.2 MB]                                                                     
Get:11 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libclang-dev amd64 1:14.0-55~exp2 [3,138 B]                                                                                     
Get:12 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 libuv1-dev amd64 1.43.0-1ubuntu0.1 [130 kB]                                                                                 
Get:13 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 llvm-14-runtime amd64 1:14.0.0-1ubuntu1.1 [484 kB]                                                                      
Get:14 http://archive.ubuntu.com/ubuntu jammy/universe amd64 llvm-runtime amd64 1:14.0-55~exp2 [3,204 B]                                                                                     
Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 llvm-14 amd64 1:14.0.0-1ubuntu1.1 [12.7 MB]                                                                             
Get:16 http://archive.ubuntu.com/ubuntu jammy/universe amd64 llvm amd64 1:14.0-55~exp2 [3,758 B]                                                                                             
Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 llvm-14-tools amd64 1:14.0.0-1ubuntu1.1 [404 kB]                                                                        
Get:18 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 llvm-14-dev amd64 1:14.0.0-1ubuntu1.1 [37.8 MB]                                                                         
Get:19 http://archive.ubuntu.com/ubuntu jammy/universe amd64 llvm-dev amd64 1:14.0-55~exp2 [1,656 B]                                                                                         
[... 98 lines cut ...]
update-binfmts: warning: python3.10 already enabled in kernel.
update-binfmts: warning: llvm-17-runtime.binfmt already enabled in kernel.
Created symlink /etc/systemd/system/multi-user.target.wants/binfmt-support.service → /lib/systemd/system/binfmt-support.service.
Setting up libclang-cpp14 (1:14.0.0-1ubuntu1.1) ...
Setting up libclang-14-dev (1:14.0.0-1ubuntu1.1) ...
Setting up llvm-14-linker-tools (1:14.0.0-1ubuntu1.1) ...
Setting up python-is-python3 (3.9.2-2) ...
Setting up llvm-14 (1:14.0.0-1ubuntu1.1) ...
Setting up llvm-14-tools (1:14.0.0-1ubuntu1.1) ...
Setting up clang-14 (1:14.0.0-1ubuntu1.1) ...
Setting up python3.10-venv (3.10.12-1~22.04.6) ...
Setting up libclang-dev (1:14.0-55~exp2) ...
Setting up pkgconf (1.8.0-1) ...
Setting up python3-venv (3.10.6-1~22.04.1) ...
Setting up clang (1:14.0-55~exp2) ...
Setting up llvm (1:14.0-55~exp2) ...
Setting up llvm-14-dev (1:14.0.0-1ubuntu1.1) ...
Setting up llvm-dev (1:14.0-55~exp2) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...
nirmalya@bluewhale:book-blinky$ 

```

So, I am not stuck. However, I thought I should bring this up, in case the book needs an update, or I am supposed to take a different step.

SergioGasquez, 2024-10-18:
Thanks for reporting it! We are evaluating how to simplify and unify the installation's instructions

debrutal, 2024-11-17:
As stated in [libudev-sys crate](https://crates.io/crates/libudev-sys) it shall be:

```shell
sudo apt-get install libudev-dev
```
not ` libuv-dev`.

Then the cargo installs work as expected.

Fix proposal https://github.com/esp-rs/std-training/pull/267

SergioGasquez, 2024-11-18:
Closed by https://github.com/esp-rs/std-training/pull/267

closed 2024-11-18 by SergioGasquez