esp-rs/std-training #266: hardware-check build fails with "unknown target riscv32imc_zicsr_zifencei-esp-espidf"

tojikomori, 2024-11-03:
The hardware-check fails to build, dying with the error:

>   error occurred: unknown target `riscv32imc_zicsr_zifencei-esp-espidf`

[Gist with full output.](https://gist.github.com/tojikomori/529f38e7c86ee0a524c143e209e98761)

Calling `rustc --print=target-list` I see `riscv32imc-esp-espidf` but not `riscv32imc_zicsr_zifencei-esp-espidf`

I also tried the docker container but to my surprise I saw the same error there too.

My host environment is x86 Linux 6.9.3, Pop OS 24.04 noble (Ubuntu derived)

Vollbrecht, 2024-11-03:
yeah that is a regression introduced by the latest release of the  `cc-rs` crate yesterday. We don't directly depend on it but do so through the `cmake` crate and thous can't pin the dependency.

In order to make it work you need to add a cc version < 1.1.32 explicitly into your binary Cargo.toml as a dependancy. That should make it work.

tojikomori, 2024-11-03:
Thanks! Adding cc 1.1.31 resolved this for now.

SergioGasquez, 2024-11-05:
Will keep this open until https://github.com/rust-lang/cmake-rs/pull/225 is merged, once merged and released, this should be fixed

faern, 2025-01-31:
Looks like https://github.com/rust-lang/cmake-rs/pull/225 is merged and released. Should this be fixed/closed?

closed 2025-01-31 by SergioGasquez